PX4/PX4-Autopilot #8746: [v1.7.x] Initial parameter loading via SiK radios in QGroundcontrol stuck

svpcom, 2018-01-22:
I've found a strange issue when initial parameter loading *sometimes* stuck when using SiK radio.
If I use another channel (for example UART for companion computer) it loads successful. When this happens it persists until controller reboot. Qgroundcontrol reboot doesn't help. I've don't seen this on PX4 v1.6.

Sik radio: tried both mavlink  v1 and v2 (from https://github.com/LorenzMeier/SiK).

```
nsh> mavlink status

instance #0:
	GCS heartbeat:	355442 us ago
	mavlink chan: #0
	type:		3DR RADIO
	rssi:		208
	remote rssi:	200
	txbuf:		99
	noise:		64
	remote noise:	54
	rx errors:	0
	fixed:		0
	flow control:	OFF
	rates:
	tx: 0.713 kB/s
	txerr: 0.000 kB/s
	rx: 0.051 kB/s
	rate mult: 0.411
	accepting commands: YES, FTP enabled: NO
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS1 @57600)

instance #1:
	mavlink chan: #1
	no telem status.
	flow control:	OFF
	rates:
	tx: 7.217 kB/s
	txerr: 0.000 kB/s
	rx: 0.000 kB/s
	rate mult: 0.511
	accepting commands: YES, FTP enabled: YES
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS2 @1500000)
```

https://logs.px4.io/plot_app?log=fc30cd9d-dbd4-45bb-9b78-aed21795e62f

screencast with qgroundcontrol:
https://drive.google.com/open?id=15QIV_hUYvJ9b_k1FzkWRM6zyYOaoDb4S

qgroundcontrol verbose logs:
[qgc.log.zip](https://github.com/PX4/Firmware/files/1653552/qgc.log.zip)

another screencast (with log capture process):
https://drive.google.com/open?id=19XxdTa-akUOsAeQBiqocP5shjNRPeGZd

dagar, 2018-01-22:
I'll try to reproduce this locally with additional instrumentation/logging. Part of it looks like QGC/PX4 param sync getting into a bad state.

dagar, 2018-01-22:
@DonLakeFlyer FYI

DonLakeFlyer, 2018-01-29:
@svpcom Make sure you are testing with a QGC daily build. There have been a large number of parameter loading fixing which have gone in since last stable.

svpcom, 2018-02-01:
@DonLakeFlyer  Use 8bf441afa01b2acfa544012947a40635e8e8a6b0. Issue still happens
[qgc.log.zip](https://github.com/PX4/Firmware/files/1684735/qgc.log.zip)

svpcom, 2018-02-01:
I've tried with RTS/CTS flow control on SiK radio - stuck still eventually happens

DonLakeFlyer, 2018-02-01:
The key log entry is here: 
```ParameterManagerLog: "V:1 C:1" Giving up on (paramIndex: 165 retryCount: 6 )```

QGC asked for that param 5 times. It never got it so it gave up. Nothing QGC can do about that from its end. All other params came across ok.

closed 2019-02-18 by stale[bot]