PX4/PX4-Autopilot #8747: Mocap follow path unstable

glennliu, 2018-01-23:
Hi 

I tried us PX4 follow a certain trajectory, with Optitrack. It is configured as below,

PX4 Firmware:  px4fmu-v2_default
Estimator: LPE
Onboard controller: Raspberry Pi
Software: Ubuntu Mate + ROS Kinetic 
Tele: Xbee 900MHz
Drone diagonal size: 300mm
Selected Airframe:DJI Flame Wheel F330 Quadrotor x (4010)

I set a trajectory to let drone to follow. A example log is [here](https://logs.px4.io/plot_app?log=a26cf12c-8248-4d76-82bf-0c6938e70b63). 
The drone cannot follow path stable. There is big drift in height, as well as XY direction. 

how to make our drone follow more stable? 
Should I modify airframe torque, moments and weight in PX4 source code? if yes, how?

glennliu, 2018-01-23:
@jlecoeur  @XingGx
I saw your module in  

> /Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py

If I change a .toml file for our drone airframe. Do you think it help its position control?

Thanks

jlecoeur, 2018-01-23:
Changing the mixers will not help here.
You have excessive vibration level (cf https://dev.px4.io/en/log/flight_log_analysis.html), check if the propellers are balanced, or add foam to isolate the fmu.
Then, if that is not enough, try to follow the PID tuning guide https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html

glennliu, 2018-01-25:
I changed a new airframe, with new motor and esc. It works much better now. I think previous one didn't choose a good power system.

jlecoeur, 2018-01-25:
Ok great to hear that. 
I close this issue, do not hesitate to re-open if needed.

closed 2018-01-25 by jlecoeur