PX4/PX4-Autopilot #8756: Snapdragon Eagle PWM: non posonsive

santiago3dr, 2018-01-24:
Running current master ae52f74 copter only responds to throttle input.
Pitch, Roll and kill switch do not look to have any affect, similar to issue #8718 

Log is not being written properly as size is 0 kb on two separate tests:
[image]

When connecting to QGroundControl, i get the following errors:
IGN REQUEST LIST: BUSY
Mission storage: Unable to read from microSD
[image]

Mission transfer failed. Retry transfer. Error: Vehicle returned error: Unspecified error (MAV_MISSION_ERROR).
[image]

santiago3dr, 2018-01-24:
@mcharleb thoughts?

santiago3dr, 2018-01-25:
quick update on this; flashed daily master 757f0e7 and the flight controlle ris now logging:
https://logs.px4.io/plot_app?log=560805eb-2d55-447c-8065-058a1b54643c

armed in my hand and the copter looks to try and stabilize itself but there is no response to pitch/roll inputs despite inputs being show correctly in qgroundcontrol

flashed eagle legacy with the qualcomm 4in1 escand that copter looks to be working just fine with all inputs being responsive and logging properly:
https://logs.px4.io/plot_app?log=181dbd36-dfc7-44d7-9404-1ab70acb991b

santiago3dr, 2018-01-30:
last flight to be resposive running c0c0666 from early January
https://logs.px4.io/plot_app?log=e148ae2e-1345-40d8-b1b2-3aa024ee1983

ChristophTobler, 2018-01-30:
Same for me with current master -> quad flipped at takeoff.
I tried reverting https://github.com/PX4/Firmware/pull/8774/files to see if this is the issue but it's not.

Last commit that I was flying with is https://github.com/PX4/Firmware/commit/45458fe9eb6fedee95621a607fd94077cdb852e0

I'll have another look at it tomorrow.

ChristophTobler, 2018-01-31:
@santiago3dr I just tested again with latest master and it works...
The issue wasn't current master for me but the parameters. I still had `EKF2_AID_MASK` set to use EV yaw and that's why I never got a valid attitude... (I had this value because I was testing with external vision).

So can you check the parameters? Maybe share them here if it's still not working.

ChristophTobler, 2018-01-31:
Ok, so the EKF never aligned and you therefore don't have an attitude. The reason seems to be that the baro isn't working https://logs.px4.io/plot_app?log=560805eb-2d55-447c-8065-058a1b54643c#Nav-Altitude-Estimate -> always zero

@santiago3dr Do you have another Snapdragon with which you could test?

Side note: you can check the attitude in QGC main window on the top right. If you tilt the drone it should show. If that's not working flying won't work.

santiago3dr, 2018-01-31:
pitch/roll nor compass heading look to be responsive on current master 2ff8139 
[image]

ChristophTobler, 2018-02-01:
@santiago3dr Yes, because the barometer of your Snapdragon doesn't seem to be working. The issue is not master, I checked that on my Snapdragon.

Do you have another Snapdragon board that you could try? Or can you retry the latest commit that worked for you (https://github.com/PX4/Firmware/commit/c0c0666d5c22a7f10163d5a76064cf019a676c73 as you mentioned above) to check if maybe your hardware is broken?

santiago3dr, 2018-02-01:
Reverted to a build from early January and it looks to be working properly; no errors flagged on qgroundcontrol; commit 5af4704a
indoor test:
https://logs.px4.io/plot_app?log=259e997d-f948-415a-9a1f-1148f80b0f49
outdoor test:
https://logs.px4.io/plot_app?log=7c087cb9-f4e5-4fd6-8c05-5750c500e256

the build:
[image]
[image]

ChristophTobler, 2018-02-01:
@santiago3dr Thanks! I tried on another Snapdragon board and there I have the same issue... https://github.com/PX4/Firmware/blob/9ac70ff6391fafad4238125e20a8eaddb24123a9/posix-configs/eagle/flight/px4.config#L12 seems to be the problem. I you comment that line it should work. It's just weird that it doesn't happen on the other board... I'll look further into it

santiago3dr, 2018-02-01:
commented said line on current master and it looks to be flying properly now:
https://logs.px4.io/plot_app?log=955035c8-cdb4-4ac7-bc64-2f937d572469

closed 2018-02-02 by dagar