PX4/PX4-Autopilot #8757: Enabling UAVCAN brakes mavlink shell

dusan19, 2018-01-25:
**Bug Report**

I am trying to use the CAN ESCs with the Pixracer flight controller. When I enable the UAVCAN through the QGroundControl I can not access the mavlink shell anymore.
I tried accessing the shell through the stable and daily built QGroundControl Mavlink Console and through the mavlink shell tool through the terminal. Also I tried the stable and master versions of the px4 firmware.

LorenzMeier, 2018-01-25:
UAVCAN consumes a lot of memory, so most likely that’s the issue. @dagar FYI

dusan19, 2018-01-25:
By the way, other parts of Mavlink work properly, in the QGroundControl I see the attitude feedback, I can load the parameters and the list of logs.

dagar, 2018-01-28:
@dusan19 as a temporary workaround try disabling anything you don't strictly need. 
 - `SYS_COMPANION 0`
 - using vmount or camera trigger?
 - frsky telemetry?
 - logging from boot or arm?

I'm looking into a few longer term solutions.

bkueng, 2018-01-29:
`SYS_FMU_TASK=0` is another option.

closed 2019-02-18 by stale[bot]