PX4/PX4-Autopilot #8758: LPE fuse uncontinuous mocap data

glennliu, 2018-01-25:
Hi 

I built a PX4+ROS+Optitrack system. The Optitrack data is transmitted through xbee 900MHz module, which has untrack period occured. The figure below showed mocap_x data

[image]

When flying, the drone cannot follow a continuous path smoothly. It always stop and moving on. That's caused by motion capture data blocked. The position data can be delayed. There could be 300ms drone cannot receive position data, and it suddenly received current position data. The drone could move rapidly. It is showed in the log above.

Since its hard to avoid mocap data lost, I want to make LPE fully trust IMU during motion capture lost. Here is my idea,
1. in mocap data lost, change LPE paramters to 100% trust IMU.
2. in mocap data lost, stop sending mocap_extra_pose to PX4. Let LPE calculate 100% based on IMU.

@jgoppert @TSC21 Can you give advice in this LPE improving issue? what's the main LPE parameter I should change.?

Thanks

glennliu, 2018-01-25:
@XingGx

glennliu, 2018-01-25:
The full log can be found [here](https://logs.px4.io/plot_app?log=5817c63b-a3ea-47ad-8cbd-b0b1cb0ac4ba)

TSC21, 2018-01-25:
Try decreasing `LPE_PN_P` and increase `LPE_VIC_P`.

TSC21, 2018-01-25:
And I also recommend you use a wifi link instead in order to increase the bandwidth and reduce the delay.

glennliu, 2018-01-25:
We need to fly swarm drone show in front audiences. Wifi also blocked by smartphone, and cause even larger delay than xbee. We didn't find a stable wifi solution yet. @TSC21

TSC21, 2018-01-25:
Why don't you use 5Ghz channels. The difference is that you won't be able to use the ESP8266 directly.

glennliu, 2018-01-25:
We used a Raspberry Pi onboard. I'm checking ESP8266, it supports 5 GHz?

TSC21, 2018-01-25:
It doesn't. But you could use another wifi bundle instead. I was refering the ESP cause PX4 support direct connection to this module.

glennliu, 2018-01-25:
I didn't find any 5G wifi bundle. Do you have recommendation? Also, if I connect ESP to PX4 UART, to send mocap data. Is that possible? maybe should change LPE setting I think.

TSC21, 2018-01-25:
Is is possible to connect the esp to uart. Anf it has nothing to do with LPE.

glennliu, 2018-01-25:
thanks. For 5G wifi bundle, any recommend? the tele issues bothered us a lot

jgoppert, 2018-11-13:
Closing as stale.

closed 2018-11-13 by jgoppert