PX4/PX4-Autopilot #8767: Feature Request: Multi distance sensor support for collision avoidance

mzahana, 2018-01-28:
Hi,

Is there a general framework for multi distance sensor support? The sensors can be of different types working on different busses. This can be very useful for obstacle avoidance implementation which I am happy to work with if such multi sensor interface is available.

I saw one issue (#7710) which seems staled for now. However, I am not sure if this is suitable for a generalization.

dagar, 2018-01-28:
Each distance sensor has an orientation field, and we could generalize the startup to let you easily start and configure as many sensors as you can fit. After that you "only" need to figure out how you'd actually like to use the data.

mzahana, 2018-01-28:
It would be great to have uORB topic that contains each sensor distance measurement and its orientation with respect to some reference  e.g. vehicle's front direction. If that could be available via Mavlink message, it would be great for offboard implementations as well.

dagar, 2018-01-28:
It's a separate instance of distance_sensor for each device (published orb multi), but it's more or less the same.
https://github.com/PX4/Firmware/blob/master/msg/distance_sensor.msg#L17

I'm not sure how you could expose multiple with the current mavlink messages. If you wanted all of them for OFFBOARD it might be a good FastRTPS project.

closed 2019-02-18 by stale[bot]