PX4/PX4-Autopilot #8782: Issues with FMUv5 board

bkueng, 2018-01-30:
This is a collection of problems I came across when bench-testing FMUv5:
- Telem2(J7): with SYS_COMPANION=57600, UART ttyS2: the attached telemetry does not work, there's no communication.
  I tried to manually start mavlink with: `mavlink start start -d /dev/ttyS2 -b 57600`. But mavlink immediately exits, w/o error output. @davids5 any ideas?
- SBUS RC is not working properly: channels are detected, but the decoding does not work correctly: the channels randomly jump in QGC.

LorenzMeier, 2018-01-30:
@bkueng Would be worthwhile to see where the MAVLink app exits. Sounds a bit like a regression.

bkueng, 2018-01-31:
I checked mavlink, and I found that it actually hangs when trying to open the UART: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L737. It does not return from that call, so it's something lower-level.

bkueng, 2018-02-01:
The NuttX config shows that Telem2 is a UART while the other telemetry ports are USART's, but their configuration is the same. So I guess it's using a different driver in NuttX.

davids5, 2018-02-08:
@bkueng - I am looking into this and have a few questions.

1) What are you looking at that suggests  TEL2 is connected to a UART?

On the schematic the TEL2 connector has signals USART3_{TX|RX|CTS|RTS}_TELEM2
This is connected to USART3. 
  The config has:
```
CONFIG_STM32F7_USART1=y
CONFIG_STM32F7_USART2=y
CONFIG_STM32F7_USART3=y
CONFIG_STM32F7_UART4=y
# CONFIG_STM32F7_UART5 is not set
CONFIG_STM32F7_USART6=y
CONFIG_STM32F7_UART7=y
CONFIG_STM32F7_UART8=y
```
Which should result in a mapping of:
```
# UART mapping on FMUv5:
#
# UART1			/dev/ttyS0		GPS
# USART2		/dev/ttyS1		TELEM1 (flow control)
# USART3		/dev/ttyS2		TELEM2 (flow control)
# UART4			/dev/ttyS3		TELEM4
# USART6		/dev/ttyS4		TELEM3 (flow control)
# UART7			/dev/ttyS5		?
# UART8			/dev/ttyS6		CONSOLE
```

LorenzMeier, 2018-02-08:
Hi David - it might have been USART2 / Telem 1

davids5, 2018-02-08:
Based in debugging Telem2(J7), what I have found is the open is blocked waiting on a DMA channel. I will have to review the usage and see why.

LorenzMeier, 2018-02-08:
That sounds like it.

RomanBapst, 2018-02-09:
@bkueng @davids5 I was looking at the SBUS problem and found the following:

- the jitter on the RC channels comes from the fact that for some reason we get some data on the dsm port (/dev/ttyS0 on PX4IO).
If you disable this line then the problem goes away
https://github.com/PX4/Firmware/blob/master/src/modules/px4iofirmware/controls.c#L235

So I think there are two questions:

1) What and why outputs data on the dsm port?
2) Should we add logic that does not allow simultaneous readout of various rc protocols (sbus, dsm, ppm, ...)

davids5, 2018-02-09:
@RomanBapst 

What is connected to the FMU?  Please describe the setup per connector.

davids5, 2018-02-09:
In looking at the schematic:

[image]

I suspect we are missing the logic to disable USART6_TX_TELEM3 when the IO is present and functional.

RomanBapst, 2018-02-09:
@davids5 I only had the usb cable (for power) and the rc receiver connected.

davids5, 2018-02-09:
I am afraid I do not have good news.

The Telem2(J7) issue is because the TX DMA of the FMU PX4IO driver (UART7_TX) uses the same stream as USART3_RX. So both can not have the DMA resource (open) at the same time[1].
We avoided this on V2 because the PX4IO driver was on USART6. I missed seeing the DMA contention when we added the IO processor to the FMUv5 with the redundant RC requirement already using USART6. 

[1] Because the TELEM RX data is async these 2 can not be multiplexed.

bkueng, 2018-05-28:
Fixed

closed 2018-05-28 by bkueng