PX4/PX4-Autopilot #8783: ROI Testing

DonLakeFlyer, 2018-01-30:
I test master ROI commands usiung master QGC. The attached plan has two ROIs set in it. The vehicle seems to respond correctly to the first ROI but ignores the second ROI. I tested both with MIS_MNT_YAW_CTL enabled and disabled. Both modes didn't correctly handle the second ROI.

[ROITesting.plan.zip](https://github.com/PX4/Firmware/files/1679732/ROITesting.plan.zip)

ndepal, 2018-01-31:
I can't reproduce this issue.
On a clean build of `posix gazebo_typhoon_h480` on 0c6280ba45a6408ade486d348bc28dbea448a31c with QGC daily as of https://github.com/mavlink/qgroundcontrol/commit/ce47bd4523e519ee17f0821f05f46cb20e597a93 both ROIs are handled as expected.

DonLakeFlyer, 2018-02-01:
Just tried again and it seems to be working now. I say "seems" because I swear even one time this test cycle it didn't do the second ROI correctly. But I couldn't reproduce, so I'm assuming I was seeing things incorrectly that one time.

closed 2018-02-01 by DonLakeFlyer