PX4/PX4-Autopilot #8795: Test and videotape ROI flight

LorenzMeier, 2018-02-01:
@PX4/testflights Could you please follow the instructions below for a structure scan flight and videotape it? We want to understand if the motion is too abrupt. Please post the log here.

DonLakeFlyer, 2018-02-01:
Let me video tape the creation of a structure scan and SITL test so folks can see how to make one. Take me a few minutes.

DonLakeFlyer, 2018-02-01:
https://drive.google.com/open?id=1o89DyvCzPWKS-7Uov7EO1vT1T9VvGzGd

The link is a movie which shows how to set up a small circular structure scan for flight testing.

The issue is that as it flies the circular pattern is is triggering images by distance the whole way through. If the vehicles jerk too abruptly as it transitions through the waypoint while taking an image, the images taken will be of low quality.

r0gelion, 2018-02-01:
Any specific platform you want us to test with?

LorenzMeier, 2018-02-01:
450 sized

santiago3dr, 2018-02-01:
Flew on current master 102c0ea without a camera and the jerk when passing through waypoints isn't too bad; log shows max roll rate around 120 and max pitch rate around 70:
https://logs.px4.io/plot_app?log=feaedc84-b4a1-4d6a-b9fb-bf0f71d2910a

That said, the copter does not face ROI but rather faces the next waypoint. Upon creating the mission, I see a setting saying the copter will point the camera using 'Gimbal yaw' and the option for 'Vehicle yaw' is grayed out:
[image]

DonLakeFlyer, 2018-02-01:
Structure Scan in current QGC will only work with gimbals that support yaw. Firmware+QGC changes coming soon to make it work with vehicles which don't have gimbal yaw. So what you saw was correct behavior.

Also given your screen shot. You aren't really running latest QGC master. But that's ok for this test.

r0gelion, 2018-02-01:
@DonLakeFlyer do you still need the videotaping?

LorenzMeier, 2018-02-01:
Could you please videotape from the ground?

DonLakeFlyer, 2018-02-01:
I think we're probbably ok with this. So that means 3.3 should be good to go with structure scan.

santiago3dr, 2018-02-01:
Downloaded new version of qgc and the yaw options are gone; copter still faces forward instead of next waypoint so we attached the camera to the side of the copter.
log:
https://logs.px4.io/plot_app?log=19142b1c-367f-49f0-ba0f-663de608f118

video:
https://drive.google.com/file/d/1-3hVQXUWjjNFCLFhPIo05BrR2c5CZexc/view?usp=sharing
https://youtu.be/uG_zGvzNI6s

(same video in both links but youtube is taking longer to process)

DonLakeFlyer, 2018-02-01:
Hmm from looking at the video it's not bad but not great either. @LorenzMeier Do you think a stabilized gimbal could handle that level of jerking?

LorenzMeier, 2018-02-04:
I believe it would and its not too aggressive given the camera was mounted to the drone body.

closed 2018-02-04 by LorenzMeier