PX4/PX4-Autopilot #8796: "RTL: Land at home" message not useful

DonLakeFlyer, 2018-02-01:
Create a mission which has RTL at end. Vehicle will spit this text message out which you then have to close in QGC. Not really helpful information. And annoying to have to close, noy helpful information. Not sure if there are other places this spiuts out.

DonLakeFlyer, 2018-02-01:
It comes up if you initiate the RTL yourself as well.

dagar, 2018-02-01:
I thought this was a good idea to alert the user of the landing, but perhaps not. I can remove it.

DonLakeFlyer, 2018-02-01:
I could see it being useful it the RTL was triggering by a failsafe. Since that is not user initiated. But if you trigger it yourself or run a mission your created with it specifically in it it's strange.

dagar, 2018-02-01:
By the time it gets to that part of the controller (navigator) it doesn't know the difference. Hypothetically, what do you think if the actual mode sent to the ground changed to LAND specifically during the landing portion of RTL?

DonLakeFlyer, 2018-02-01:
Does the failsage trigger some sort of other message? If so I think that would be good enough.

dagar, 2018-02-01:
There's the message for the failsafe itself we should have either way, but I'm talking specifically about the last leg of RTL when the vehicle actually starts landing. RTL could be configured to loiter indefinitely over home, for a fixed period, or switch to land immediately. Having some mechanism to alert the operator the actual landing has started is valuable.

DonLakeFlyer, 2018-02-01:
Seems reasonable to me.

dagar, 2018-02-01:
It's potentially a bit weird because the vehicle itself is not actually in LAND mode, but I don't think the distinction actually matters. @antiheavy what do you think?

The landing portion of a mission could do the same thing.

Antiheavy, 2018-02-03:
There are 2 issues being discussed here - here is my 4 cents:

**1)  "Land" mode for final phase of landing:** 
YES, I think the vehicle should switch to some form of Land mode any time it is AUTO landing.  The Mode should been seen to change in QGC and the switch should be both audible and visual.  The mode switch (all most switches IMHO) should also be logged as an event in the vehicle log file.  This includes mission landing, Return mode landing, Land command, and anything else.  It should only switch when on the final phase of whatever the situation is as @dagar describes.

**2) RTL "text messages"**
There absolutely should be both audible and visual indication and events logged in the log file for all mode switches.  Especially switches into Return mode.  There is lots of room for improvement though.

The "text messages" that pop-up and must be dismissed can sometimes be annoying, both because you must dismiss them and because the wording is often terrible.  However, the fact that this mechanism exists is a major reason we are using PX4 vs other open source options because we need the autopilot to be able to communicate problems and status to the user.  Even though users SHOULD have their volume on, many turn it off.  Having better, less annoying messages overall would help prevent people from turning off the volume on purpose.  Either way, there must be audible and visual cues of mode changes - if you don't want text messages for some of these then maybe a better QGC animation or very obvious visual indicator of mode switch is required - more than just the small text changing at the top of the screen.

There is already a mechanism for different types of text messages right?  I believe some show the pop-up and some don't?  Is it possible for a type of message to have a pop-up that automatically ages out (.e.g. 3 seconds or something)? Or maybe all messages come with a display time that lets QGC auto dismiss it after the specified time (fade away animation?) This way QGC could know if it should: read and/or display the message for XX seconds.  Maybe set display time to 0 seconds to require the user to dismiss it.

closed 2019-02-18 by stale[bot]