PX4/PX4-Autopilot #8803: No timestamp for SMBUS battery_status topic

tdnet12434, 2018-02-02:
Hi all,
The time stamp on this line would be reset to zero after voltage is updated.

https://github.com/PX4/Firmware/blob/00a47ba542b23f112e2e467a8fd5c3a1c8c322a9/src/drivers/batt_smbus/batt_smbus.cpp#L644

I also doubt about battery_status from smbus is lower rate than ADC reading and not properly sent to mavlink.

dagar, 2018-02-03:
Would you like to submit a PR fixing it? You could zero the struct above (https://github.com/PX4/Firmware/blob/00a47ba542b23f112e2e467a8fd5c3a1c8c322a9/src/drivers/batt_smbus/batt_smbus.cpp#L641) and drop the memset.

Could you expand on the problem with ADC reading? If you make the timestamp fix and generate a simple log it might help.

dagar, 2018-07-24:
This has been fixed.

closed 2018-07-24 by dagar