PX4/PX4-Autopilot #8805: Snapdragon power module not working on all boards

ChristophTobler, 2018-02-02:
The power module (ltc2956) is currently not working on all boards. Could the issue be a hardware revision?

I tried different power modules but the reason seems to be the board. It's the same OS version on all devices...

mini-dm output
```
[08500/02]  43:36.192  HAP:61:starting LTC2946  0168  df_ltc2946_wrapper.cpp
[08500/00]  43:36.192  HAP:61:started LTC2946  0176  df_ltc2946_wrapper.cpp
[08500/00]  43:36.192  HAP:61:1205568 LTC2946: Open the device path: /dev/iic-3   0111  LTC2946.cpp
[08500/02]  43:36.192  HAP:61:I2C_Write failed: 285212872 num_bytes_written 0  0414  i2c.c
[08500/00]  43:36.192  HAP:61:1205683 Error: i2c write failed. Reported -1 bytes written   0240  I2CDevObj.cpp
[08500/00]  43:36.192  HAP:61:1205712 ltc2946: configure failed!   0091  LTC2946.cpp
[08500/00]  43:36.192  HAP:61:1205735 error: sensor initialization failed, sensor read thread not started   0125  LTC2946.cpp
[08500/00]  43:36.192  HAP:61:1205768 error: Failed to start ISL   0139  LTC2946.cpp
[08500/03]  43:36.192  HAP:61:LTC2946 start fail: -1  0114  df_ltc2946_wrapper.cpp
[08500/03]  43:36.192  HAP:61:DfLtc2946Wrapper start failed  0182  df_ltc2946_wrapper.cpp
```

@mcharleb @jywilson any ideas?

karliss, 2018-12-05:
Did you tested it with power connector connected to Qualcom ESC or APM board, with battery or 5v power supply.

Driver currently expects 2 LTC2946 modules on i2c bus and will fail either in initialization or sensor reading of both of them are not present.

from my tests:
* Qualcom ESC - single LTC2964 with address 0x6a(d4), without modifications current driver fails during initialization because it tries to configure LTC2946_I2C_ADDRESS=LTC2946_I2C_ADDRESS_5V
* APM with 5v plug - 2 LTC modules but only one of them is powered, config using 0x6b(d6) succeeds  but _measure fails due to second LTC2964 not being powered
* APM with 5v and battery - everything works, although config has been applied to only one of the two LTC2964
* APM with battery - same as APM with battery and 5v plug

karliss, 2018-12-07:
Slightly related, I observed that sometimes when LTC2946 wasn't working properly it caused problems for other i2c drivers on the same bus(df_bmp280_wrapper).

closed 2019-03-21 by stale[bot]