PX4/PX4-Autopilot #8812: board manifest ideas

dagar, 2018-02-03:
@davids5 I'd like to start thinking about the board manifest idea. Let's try to gather some simple requirements.

Each board (and particular configurations) not knowing which sensors they actually have, or even how many of each it should have is a huge pain point.

It's not clear to me how far we could/should take this, but I'd like to first think about the aspects that pertain to sensors.

 - json of similar board definition
 - sensor configuration by bus
  - i2c
     - internal/external
  - spi
     - internal/external
     - chip selects
     - interrupt configuration?
  - uarts

From this we could fill in large portions of the board_config.h and somehow generate the actual startup. I think we also need some notion of required devices and optional. The board manifest could contain enough information to determine the device id. Other code could independently verify that all required sensors are present and working. Alternatively the actual startup could somehow be tied together.

How do we handle extremely variable configs like the px4fmu-v2? A subset of this configuration needs to work across all supported platforms. For example on Linux we should know which SPI, I2C, serial are available to PX4, including any additional metadata (like internal/external).

ryanjAA, 2018-02-04:
I like the idea, we've gotten all the board designs done and specific sensor routing finished and to their respective pins utilizing an F4 and power requirements for what a pretty diverse node will look like including everything listed above. Time to rid i2c... What is left open is the software implementation and what you have noted above. Once a few more details on the software are ironed out we can have some more samples made outside of what we have now.

robert-b, 2018-02-13:
the idea is nice.
at the end 'c' structure definitions should be spit out by some sort of conversion tool.
i think it is easy to convert a 'c' structure like:
typedef struct
{
	int16_t      Pin;
	int16_t      PinSource;
} Port_t;
typedef struct
{
	SPI_TypeDef *      SPIx;
	GPIO_TypeDef *     Port;
	Port_t             P[3];
	bool               Used;
	uint8_t            af;
} SPIPortDef;
into some sort of json definition.
in case dma utilisation is required the structure will grow.

the board manifest needs to be complete:
 - crystal used
 - pll variables
 - clock deviders
else the board is in a not defined state.

in case of linux i think it is not the goal to define a board support package.
but here we require a sw interface to talk with the kernel.
btw - what kernel - one with realtime extensions?
and i have never looked at the current px4 linux support ...

dagar, 2018-02-13:
Perhaps we should assemble a few examples for common boards? 
 - px4fmu-v4 or mindpx
 - a common linux board like the raspberry pi

It's not clear to me how deep into NuttX we should bother going. The primary motivation from my perspective is defining all the board specific pieces that are actually needed in PX4 (primarily for sensor drivers).

If we consumed the entire NuttX board.h and board_config.h how would that help us?
https://github.com/PX4/Firmware/blob/master/platforms/nuttx/nuttx-configs/px4fmu-v4/include/board.h
Looking through it I see a lot of commenting and common sections that wouldn't need to exist, a few critical configuration parameters (crystal, pll), and then more generic mapping.

I want to first focus on solving the core PX4 issue (board manifest for sensor drivers), but I don't see why we couldn't fairly easily describe an entire NuttX board this way.

davids5, 2018-02-13:
@dagar correct it is the PX4 sensor usage. NuttX board is compile time driven (as opposed to runtime driven like libopencm3) I do not think we want expand the scope to that level.

closed 2019-02-11 by stale[bot]