PX4/PX4-Autopilot #8813: drivers device instances

dagar, 2018-02-03:
We should keep lists (probably per bus) of all the [Device](https://github.com/PX4/Firmware/blob/master/src/drivers/device/Device.hpp) (I2C/SPI/CAN) instances in the system.

 - this would make deconfliction easier and explicit
 - allow us to drop unnecessary CDev usage (/dev/mpu6000_accel, /dev/mpu6000_accel1, /dev/mpu6000_accel_ext, /dev/mpu6000_accel_ext1, etc https://github.com/PX4/Firmware/blob/master/src/drivers/mpu6000/mpu6000.h#L52)
 - improve the debug tooling/reporting
    - print a simple summary of all drivers by bus, address, configuration, driver type
 - it might make sense for serial drivers (eg most lidar, px4io) to become Devices as well

@davids5

TSC21, 2019-09-23:
Still relevant

dagar, 2020-04-08:
Done!

closed 2020-04-08 by dagar