PX4/PX4-Autopilot #8815: ROS leaking in as dependency

LorenzMeier, 2018-02-03:
@dagar We need a CI system set up without any ROS to avoid ROS packages leaking in as unintended dependency:

```
Traceback (most recent call last):
  File "/Users/lorenzmeier/src/Firmware/Tools/sitl_gazebo/scripts/jinja_gen.py", line 9, in <module>
    import rospkg
ImportError: No module named rospkg
```

@TSC21 FYI

LorenzMeier, 2018-02-03:
@dagar And Gitmate starts to fail us badly - can we disable related issues? It has no clue what it's doing.

LorenzMeier, 2018-02-03:
Just disabled duplicate checking.

TSC21, 2018-02-03:
This will be fixed on the container. This is related with the jinja generator. Will fix it today.

LorenzMeier, 2018-02-03:
Fixed in df32297

@TSC21 This failed on my machine. So you don't need to fix it on the container. Instead we need a new container *without ROS* to make sure we're not picking up dependencies we don't want in Firmware.

closed 2018-02-03 by LorenzMeier

TSC21, 2018-02-03:
The question is that you can install `rospkg` without ROS. That'a why I added to the install instructions on `sitl_gazebo` side.

LorenzMeier, 2018-02-03:
That's not a good argument. It still becomes a dependency. If you start thinking that way you will have soon hundreds of packages as dependencies and the only working platform will be Ubuntu.

Instead, any package pulled into the system must serve a purpose. If it doesn't (there is no value for non-ROS enabled Gazebo SITL) it must not become a dependency in the first place.

And to enforce all this we need a minimal image running CI.

TSC21, 2018-02-03:
I understand. I actually initially was for not adding any ROS dependency but having it available by `pip` did change my mind. But that makes sense

dagar, 2018-02-04:
gitmate silenced (it's still going to be responsible for closing stale issues).