PX4/PX4-Autopilot #8834: batt_smbus is broken

AlexKlimaj, 2018-02-07:
Trying to compile batt_smbus in the latest master and there are numerous compile errors. 

Can we bring this up to date and make it more generic by removing the 3dr solo information?

dagar, 2018-02-07:
Surprisingly it's not included in any configs. Would you like to post the errors to work through getting it working again? Once it's working we can add it to the larger build targets (px4fmu-v3, v4, etc).

AlexKlimaj, 2018-02-07:
Yeah, here is the first one. 

> ../../src/drivers/batt_smbus/batt_smbus.cpp: In member function 'int BATT_SMBUS::write_reg(uint8_t, uint16_t)':
../../src/drivers/batt_smbus/batt_smbus.cpp:749:31: error: 'debug' was not declared in this scope
   debug("Register write error");

And the second one

> ../../src/drivers/batt_smbus/batt_smbus.cpp: In member function 'uint8_t BATT_SMBUS::get_PEC(uint8_t, bool, const uint8_t*, uint8_t) const':
../../src/drivers/batt_smbus/batt_smbus.cpp:846:44: error: passing 'const BATT_SMBUS' as 'this' argument discards qualifiers [-fpermissive]
  tmp_buff[0] = (uint8_t)get_device_address() << 1;

I got it to compile by commenting out the debug calls, GETPEC function, and its calls.

dagar, 2018-02-07:
Ok, this looks pretty straightforward to recover.

dagar, 2018-02-07:
@AlexKlimaj do you want to open a pull request adding this to the board you care about and fixing the first few issues? From there I'll jump in and fix the remaining failures.

AlexKlimaj, 2018-02-08:
I started working on it yesterday. It seems that the UORB battery messaging has changed since it was last used. I am also writing it for a TI BQ40Z50-R1/R2.

dagar, 2018-02-08:
Please give https://github.com/PX4/Firmware/pull/8840 a try. I believe it will publish to the 2nd instance of battery_status (battery_status1), so we may need to add something to disable/ignore the default ADC battery status handling.

AlexKlimaj, 2018-02-08:
I can try it today. I found yesterday that setting BAT_SOURCE to 1 will set the battery ADC's to be ignored. It's in sensors.cpp.

dagar, 2018-02-08:
Ah, right BAT_SOURCE. That's a bit of a hacky workaround, but we can properly expand the usage of that param to work correctly here.

closed 2018-02-08 by LorenzMeier

AlexKlimaj, 2018-02-08:
I've created https://github.com/PX4/Firmware/pull/8854

PR #8840 (2018-02-08): batt_smbus readd to configurations and fix compilation errors
@AlexKlimaj could you test that this actually works? I don't have one.

 - fixes #8834