PX4/PX4-Autopilot #8837: Make px4fmu-v2_default

ismatouu, 2018-02-07:
Hello,
Any help is appreciated 
[1/3] Linking CXX executable nuttx_px4fmu-v2_default.elf
FAILED: : && /home/ismat/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-g++   -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG  -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Wl,--warn-common,--gc-sections platforms/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/__/__/src/platforms/empty.c.obj  -o nuttx_px4fmu-v2_default.elf -L/home/ismat/Firmware/build/px4fmu-v2_default/external/Install/lib -Wl,--script=/home/ismat/Firmware/build/px4fmu-v2_default/NuttX/nuttx/configs/px4fmu-v2/scripts/ld.script -Wl,-Map=nuttx_px4fmu-v2_default.map -Wl,--warn-common,--gc-sections -Wl,--start-group msg/libuorb_msgs.a src/platforms/common/libplatforms__common.a src/drivers/differential_pressure/ets/libdrivers__ets_airspeed.a src/drivers/differential_pressure/ms4525/libdrivers__ms4525_airspeed.a src/drivers/differential_pressure/ms5525/libdrivers__ms5525_airspeed.a src/drivers/differential_pressure/sdp3x/libdrivers__sdp3x_airspeed.a src/drivers/distance_sensor/ll40ls/libdrivers__ll40ls.a src/drivers/distance_sensor/mb12xx/libdrivers__mb12xx.a src/drivers/distance_sensor/sf0x/libdrivers__sf0x.a src/drivers/distance_sensor/sf1xx/libdrivers__sf1xx.a src/drivers/distance_sensor/srf02/libdrivers__srf02.a src/drivers/distance_sensor/srf02_i2c/libdrivers__srf02_i2c.a src/drivers/distance_sensor/teraranger/libdrivers__teraranger.a src/drivers/distance_sensor/tfmini/libdrivers__tfmini.a src/drivers/distance_sensor/ulanding/libdrivers__ulanding.a src/drivers/airspeed/libdrivers__airspeed.a src/drivers/barometer/ms5611/libdrivers__ms5611.a src/drivers/boards/common/libboards_common.a src/drivers/boards/common/stm32/libboard_stm32.a src/drivers/boards/px4fmu-v2/libdrivers__boards__px4fmu-v2.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/device/libdrivers__device.a src/drivers/gps/libdrivers__gps.a src/drivers/l3gd20/libdrivers__l3gd20.a src/drivers/led/libdrivers__led.a src/drivers/lsm303d/libdrivers__lsm303d.a src/drivers/magnetometer/hmc5883/libdrivers__hmc5883.a src/drivers/magnetometer/lis3mdl/libdrivers__lis3mdl.a src/drivers/mpu6000/libdrivers__mpu6000.a src/drivers/mpu9250/libdrivers__mpu9250.a src/drivers/pwm_input/libdrivers__pwm_input.a src/drivers/pwm_out_sim/libdrivers__pwm_out_sim.a src/drivers/px4flow/libdrivers__px4flow.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/px4io/libdrivers__px4io.a src/drivers/rgbled/libdrivers__rgbled.a src/drivers/stm32/libdrivers__stm32.a src/drivers/stm32/adc/libdrivers__stm32__adc.a src/drivers/stm32/tone_alarm/libdrivers__stm32__tone_alarm.a src/drivers/vmount/libdrivers__vmount.a src/modules/sensors/libmodules__sensors.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/systemcmds/ver/libsystemcmds__ver.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/events/libmodules__events.a src/modules/land_detector/libmodules__land_detector.a src/modules/load_mon/libmodules__load_mon.a src/modules/mavlink/libmodules__mavlink.a src/modules/navigator/libmodules__navigator.a src/modules/ekf2/libmodules__ekf2.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/fw_pos_control_l1/launchdetection/liblaunchdetection.a src/modules/fw_pos_control_l1/runway_takeoff/librunway_takeoff.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/mc_att_control/tailsitter_recovery/libtailsitter_recovery.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/logger/libmodules__logger.a src/modules/dataman/libmodules__dataman.a src/modules/systemlib/libmodules__systemlib.a src/modules/systemlib/param/libmodules__systemlib__param.a src/modules/uORB/libmodules__uORB.a src/lib/controllib/liblib__controllib.a src/lib/conversion/liblib__conversion.a src/lib/DriverFramework/framework/liblib__DriverFramework__framework.a src/lib/ecl/liblib__ecl.a src/lib/geo/liblib__geo.a src/lib/geo_lookup/liblib__geo_lookup.a src/lib/led/liblib__led.a src/lib/mathlib/liblib__mathlib.a src/lib/mixer/liblib__mixer.a src/lib/tunes/liblib__tunes.a src/lib/version/liblib__version.a platforms/nuttx/src/px4_layer/libplatform_px4_layer.a NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libxx/libcxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc ROMFS/libromfs.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a && :
/home/ismat/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: nuttx_px4fmu-v2_default.elf section `.data' will not fit in region `flash'
/home/ismat/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 2369 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

taileron, 2018-02-07:
GCC above 6.3.1 optimises the memory usage while compiling as I learned ...
GCC 7.2.1 is now recommended.
see #8773

ismatouu, 2018-02-08:
@taileron so, the GCC 5.4 will not build the px4fmu-v2_default ..?

taileron, 2018-02-08:
current master still generates an error: overflow by 3445 bytes for v2_default with GCC 5.4.1
without the GCC memory optimisation feature it needs to remove unused modules to build successful
GCC 7.2.1 takes this feature for reducing memory usage to fit into small flash size

vuanhduy, 2018-03-06:
I have just compiled the code myself and it was fine. Did you adjust something in the px4fmu-v2_default file? the reason you got that error is to have the px4 image file bigger that 1 MB. In other words, you can only install limited number of modules due to the small flash memory of your flight controller.

taileron, 2018-03-06:
... exactly that´s it, but GCC 7.2.1 uses a memory optimisation feature that saves space to integrate more modules if necessary. Currently both GCC (5.4.1 or 7.2.1) can be used to compile PX4.
I´m copying the desired GCC folders into a fixed system path for fast changing between the GCC versions cause I need an additional one for compiling e.g. Betaflight.

taileron, 2018-03-08:
with the current master it works with both GCC versions, tried it now -> solved

huangwen0907, 2018-04-04:
I have a similar problem with this
>/home/hw0907/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102214 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/home/hw0907/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: nuttx_px4fmu-v2_default.elf section `.ARM.exidx' will not fit in region `flash'
/home/hw0907/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102214 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/home/hw0907/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102214 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/home/hw0907/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 12245 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

hamishwillee, 2018-04-11:
This error means that the flash is too big (>1MB). As of this morning the build worked fine IFF **you use the supported compiler**. However as the firmware is so close to the limit, it may break again "somewhat randomly". 

The issue is documented in the building section for this target: https://dev.px4.io/en/setup/building_px4.html#building_nuttx

I am therefore closing.

closed 2018-04-11 by hamishwillee