PX4/PX4-Autopilot #8842: Changing magnetometer rotation doesn't affect calibration

nanthony21, 2018-02-08:
I've been getting the Preflight Fail: Mag Sensors Inconsistent error on a Pixracer running version 1.7.3. I followed the directions I found here: #8098 to log the raw data from each of the sensors. Upon viewing the logs I saw that while the Z readings are ok, the X and Y reading of MAG0 are opposite those of MAG1 and MAG2 (This suggests that the yaw rotation is off by 180 degrees). However, after setting MAG0's rotation to yaw180 and going through calibration again the log showed to exact same issue. I've tried No Rotation, Yaw180, Yaw90, and even Roll90, but the logs always show the X and Y readings of MAG0 being 180 degrees out of phase while the Z reading is fine.

No Rotation: https://logs.px4.io/plot_app?log=97859587-8eb1-4ae3-ae51-62fc19194d9e
Yaw 180: https://logs.px4.io/plot_app?log=b134fd31-ba86-4f47-975b-e8712e964ba4
Yaw 90: https://logs.px4.io/plot_app?log=bcbc062f-1510-4476-a84b-34d0bd693cea
Roll 90: https://logs.px4.io/plot_app?log=143dad2b-a215-4119-b5c5-77f486d5becf

ndepal, 2018-02-08:
I have noticed this too. The data logged in `sensor_mag` are not rotated according to the parameter.
This is despite rotation apparently taking place before this message is published: https://github.com/PX4/Firmware/blob/master/src/drivers/magnetometer/hmc5883/hmc5883.cpp#L1007

But apparently the rotation is only really applied in https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L786

To check the rotations are right, you can use `sensors status` and compare different mags in a couple of different rotations.

nanthony21, 2018-02-09:
Thanks. Using the `sensors status` nutshell command I was able to verify that the correct orientation was my  original setting of "No rotation". However I still get the Preflight Fail message. On the release version of the firmware I didn't have access to the `COM_ARM_MAG` parameter. I updated the firmware to the development version and changed `COM_ARM_MAG` from 0.15 Gauss to 0.30 Gauss and now the issue appears to be resolved.

I would guess that the reason I was seeing the external mag disagree with the internal mags by a yaw of 180 degrees even though the appropriate yaw for my external mag is 0 degrees is that the pixracer board orientation is set to yaw 180. This suggests that the optional sensor datalogging applies the board rotation to internal sensors as it should, but fails to apply the rotation that is set for external sensors.

closed 2019-02-11 by stale[bot]