PX4/PX4-Autopilot #8843: Terrain estimation validity broken

mhkabir, 2018-02-08:
Something seems to have recently broken the `dist_bottom_valid` field in EKF2. It is no longer valid when it should be : 

To reproduce, run any model with the laser in SITL. e.g `make posix_sitl_default gazebo_iris_irlock`, and then run `listener vehicle_local_position` :
```
TOPIC: vehicle_local_position instance 0 #1
timestamp: 60855301
xy_valid: True
z_valid: True
v_xy_valid: True
v_z_valid: True
x:  -0.1160
y:  -0.0247
z:  -1.5499
delta_xy:   0.0023  -0.0068 
xy_reset_counter: 1
delta_z:   0.0000
z_reset_counter: 0
vx:   0.0221
vy:   0.0926
vz:  -0.0076
z_deriv:  -0.0018
delta_vxy:   0.0073  -0.0143 
vxy_reset_counter: 1
delta_vz:  -0.0262
vz_reset_counter: 1
ax:   0.1636
ay:   0.1043
az:   0.0019
yaw:   0.0512
xy_global: True
z_global: True
ref_timestamp: 10266664
ref_lat:  47.3977
ref_lon:   8.5456
ref_alt: 488.1593
dist_bottom:   1.8302
dist_bottom_rate:   0.0076
dist_bottom_valid: False
eph:   0.4216
epv:   0.1440
evh:   0.1759
evv:   0.0500
vxy_max:   0.0000
limit_hagl: False

```

@ChristophTobler @priseborough

ChristophTobler, 2018-02-08:
The issue is not related to ecl. The issue is that `iris_irlock` uses the lidar model and https://github.com/PX4/sitl_gazebo/blob/master/include/gazebo_mavlink_interface.h#L103 has been changed to listen to `sf10a` (which is wrong IMO). I'll push a fix to sitl_gazebo.

ChristophTobler, 2018-02-08:
I shouldn't have closed the issue before the fix is in...

ChristophTobler, 2018-02-08:
fixed in https://github.com/PX4/Firmware/pull/8848

closed 2018-02-08 by ChristophTobler