PX4/PX4-Autopilot #8844: mc_pos_control: att_sp is not as smooth for V1.7 code as it is for V1.6.5 because of the value of manual_stick_input changing.

guduzhe123, 2018-02-08:
My multirotor shakes while it is braking in position mode. 
I simulated in gazebo using "make posix gazebo_typhoon_h480" and this phenomenon happened again.  I find that the V1.7 adds a new value named "manual_stick_input" and it influences the attitude set-point.

enum manual_stick_input {
		 brake,
		 direction_change,
		 acceleration,
		 deceleration
	};
When the intention changes from "deceleration" to "brake", the att_sp has a sharp change. That makes the multirotor  quite unstable. The simulation is here: the "M1:.x" is the intention of xy direction. 

[image]

Log is here: https://logs.px4.io/plot_app?log=dcde664a-bc3c-4372-a537-d2e8c6cd4102

[image]
[image]

tops4u, 2018-02-08:
Just saw your Issue. This may be related #8847

guduzhe123, 2018-02-09:
I tried to adjust some new added parameters: "MPC_JERK_MAX", "MPC_JERK_MIN" and "MPC_DEC_HOR_SLOW", and also adjusted  PID, but this behavior still happened.

tops4u, 2018-02-09:
Yes in may case it also still happened, but the Parameter lowered the Phenomenon to an acceptable degree. I think the was introduced in 1.7.0-rc0 when the Jerk Parameter was introduced. Just as you I had no such Issues prior to 1.7.

Stifael, 2018-02-09:
@guduzhe123 that is correct that there will be a quick jerk when braking. The reason is that the slewrate is reset, otherwise it might be possible that the vehicle continues to accelerate in the wrong direction until velocity setpoint and actual velocity caught up. To prevent the jerking, a jerk parameter was introduced.
Please see [here](https://github.com/PX4/Firmware/pull/7242#issuecomment-321201103) for more info. I am also working on a documentation for all the parameters. It probably also makes sense to have jerk set as default.

closed 2019-01-07 by guduzhe123