PX4/PX4-Autopilot #8847: MC: JERK Param causing strange behaviour in PosCTL

tops4u, 2018-02-08:
**Bug Report** 

I have had some issues after starting to use 1.7.0 Firmware. At first I thought this was because I have started using newer ESC (more aggressive by actively braking the Motors - BLHeli Damped Light), but this seems not be the case. After making the ESC behaving like to ones before the Vehicle still behaved the same having odd "hickups" when braking in PosCTL (Controller Switchover from Velocity to Position Controller). Thanks to @dagar I have been able to make it almost behave as it was prior to 1.7.0 by setting MPC_JERK_MAX to a Value of 5.0. 

As far as I understood when this Param is set to 0 then MPC_ACC_HOR_MAX kicks in giving the controller unlimited Rate authorization in order to reach this value. This caused steep Setpoint Steps in my case (caused by puting the Stick back to center, this is a soft case, I have seen larger Steps in the Setpoint when Controllers switch):

[image]

I was not able to controll this behaviour by changing the MPC_XY_VEL PID Values or the MPC_XY_P Value. After setting the JERK Value to 5 this will limit the Rate change and it behaves as befor in PosCTL Mode. Yet in MIssion Mode I do still see Steps in the Setpoint:

[image]

Expected behavior:
- Default MPC_JERK_MAX to a Default > MPC_JERK_MIN to limit the Setpoint JERK when braking.
- Check why there are still Setpoint Steps in Mission Mode and how to avoid this. 

Log (Default Values for PosCTL PID & JERK): https://review.px4.io/plot_app?log=428bd277-746b-4697-85c6-316d14288a15
Log (Lower PosCTL PIDs and JERK = 5.0): https://review.px4.io/plot_app?log=2c9f2e09-9794-4135-aeb3-b50ec2408a73
Log (Mission Flight with Setpoint Steps): https://review.px4.io/plot_app?log=63bd8007-cd57-4638-b811-aacd7dcf409f

Stifael, 2018-03-12:
@tops4u just a few comments.
The smoothing is only used in auto mode and hence as no effect for any auto related maneuver.
If the flightbehaviour is weird, then I would not to change the smoothing and lower lever controller gains simultaeneously, but rather would turn off any of the smoothing to see if the tracking is good. The way you can do that is as follow:
- set MPC_JARK_MAX < MPC_JERK_MIN. This will turn off any jerk smoothing during braking and is also set by default. 
- set all acceleration parameter to large values:
MPC_ACC_HOR_MAX, MPC_ACC_HOR, MPC_DEC_HOR_SLOW, MPC_ACC_UP_MAX, MPC_ACC_DOWN_MAX to 1000 or similar.
If you fly now manual position controlled mode, the vehicle should track all the setpoint well. The tracking should be very tight, othewise you need to tune the PID-gains indeed. 
If the tracking is tight, then you can start adding back some of the smoothing stuff which adjust the sepoint (vs PID gain, which changes the tracking of the setpoint).

For the mission part, I commented here  #9051.

tops4u, 2018-03-12:
Hi @Stifael thanks for you comment. 

I thought the Tracking is good, that is why when there is the Step from Velocity Controller to Position Controller It really looked rather scary in the beginning since the Setpoint is performing a Step. 
So when Jerk Param smoothes out the Controller in PosCTL Mode, why not also apply it in Auto Modes? As currently PX4 does not support slopes for ex. at a rectangular Waypoint it will almost come to a full stop just to start accelerating on the other axis. If Setpoint behaviour is like in PosCTL (I assume the AutoMode is just the Input of the PosCTL -> Attitude Controller Cascade) this looks rather ugly. 

This happens, when I just slowly set the Pitch to Center (0). IMHO this will always JERK a lot and twice.
[image]

For the eye, flying PosCTL and Auto modes looked much smoother prior to 1.7 Release. So either the requirements to have a really thight PID setting have been augmented (probably beyond my skills) or I apply it wrong, or whatever. I'm a bit lost with this.

Stifael, 2018-03-12:
> So when Jerk Param smoothes out the Controller in PosCTL Mode, why not also apply it in Auto Modes

The jerk params is only active during braking in manual controlled mode and off by default. Set MPC_JERK_MAX to something larger than 15 to see how it effects the braking. For smoothing during non-brake maneuvers but still manual controlled, you can play around with all the different acceleration parameters.

About auto. The problem with smoothing is that it changes the setpoint, and not the tracking. In auto, the goal is to follow the track as good as possible, which cannot be achieved with smoothing and also the main reason why the logic changed. However, I agree that the rather strong jerk at the waypoints are annoying, but that is more a problem of the trajectory generation than smoothing. 

Anyway, since the logic for generating setpoints is different for auto and manual controlled mode, it makes sense to keep the issues separated. 

For auto related issue, lets use the discussion here #9051 and keep this issue for manual controlled mode.

tops4u, 2018-03-12:
Still I would propose to set JERK to a default in order not to scare off New Users or Switchers from < 1.7.0 to newer releases. And/or probably document the Feature a bit better. 

I could do a PR if this is a way to go.

LorenzMeier, 2018-03-12:
I think that’s a fair point. @thomasgubler FYI

Stifael, 2018-03-13:
> Still I would propose to set JERK to a default 

Sure. 
In the future what I would like to have is to have all the smoothing turned off, just leaving the pure position/velocity control. the user can then go through a UI tutorial that helps him to achieve the desired smoothness. Otherwise it is just to many things that possibly can go wrong.

> I could do a PR if this is a way to go.
Thanks. But I don't know what a good default value could be. 

> And/or probably document the Feature a bit better.
Ok, I will try to add that stuff to the dev guide / user guide. I actually started it, but never finished it unfortunately.

tops4u, 2018-03-15:
I have to come back to this @Stifael. I had my maiden Flight with the VTOL yesterday, and after backtransition I get a hughe Spike in the Pitch Setpoint:

[image]

I asume this results in the Forward Movement from the Transition which the PosController would like to compensate for, and it looks a bit scary. Is this related to this issue when it switches from Velocity to Position?

Log is here: https://review.px4.io/plot_app?log=f38b2d9e-252d-4e40-b4fb-5f9cdbc02784 using Defaults for Quadplane Ranger EX

tops4u, 2018-03-15:
Just noted, that the Jerk Param is already in Place. Maybe I should just make the Transition Phase longer in order to brake the Plane further down before it switches to PosCtrl.

Stifael, 2018-03-15:
I started now to add documentation to the devguide. 
https://github.com/PX4/Devguide/pull/484/files

tops4u, 2018-05-03:
@Stifael thanks for the explanation in the Docs, but IMHO this is not a tuning guide. It explains what the parameters are doing, and I think it is a good thing. But some lines about how to approach Tuning of those would be handy.

closed 2019-02-11 by stale[bot]