PX4/PX4-Autopilot #8850: pwm app bug

RomanBapst, 2018-02-08:
When using the pwm app with "test" argument, e.g.

```
pwm test -c 1234 -p 1500
```

you do not get a clean pwm signal with only 1.5 ms pulses but you actually get some disarmed pulses (0.9 ms) in between. I noticed this issue when trying to actuate a DJI FOC ESC. The signal would just make the motor vibrate and make some noise.
I guess you do not notice this issue with cheaper ESC's.

@bkueng FYI

ndepal, 2018-02-08:
Last time I used `pwm test` killing `mc_att_control` first helped. Otherwise I had similar problems.

RomanBapst, 2018-02-08:
@ndepal Thanks for the feedback! I'll give it another try and see if this helps. I'm just thinking if we could document this somewhere, at least print it when using the help argument if this is not already done.

AlexKlimaj, 2018-02-09:
I had this same problem last week when trying to test motor/esc/battery current draw. I was unable to get pwm test working even after I killed mc_att_control.

I scoped the PWM output on the board and found that my commanded test pulse width was getting sent, but with periodic single disarm pulses causing the motors to stutter.

hamishwillee, 2018-02-09:
> @ndepal Thanks for the feedback! I'll give it another try and see if this helps. I'm just thinking if we could document this somewhere, at least print it when using the help argument if this is not already done.

@RomanBapst @bkueng Yes. The PWM modules is documented in-source and that is extracted here: https://dev.px4.io/en/middleware/modules_command.html#pwm

So perhaps worth adding as a note against the `test` argument?

RomanBapst, 2018-02-09:
@hamishwillee We got some additional feedback and will first investigate if there is indeed a bug that mixes pwm pulses even if attitude controllers are shut off.

RomanBapst, 2018-02-09:
[image]

JohnSnowball, 2018-02-09:
This exists for both motor_ramp and pwm test commands, they work well on pwm_output0, as for pwm_output1, there's mixed signal, one is given by me, other one is 900us.
All att control modules are shut off.

dagar, 2018-02-12:
@RomanBapst what's the full setup? I'm wondering if this is specifically a px4fmu issue.

AlexKlimaj, 2018-03-15:
Still seeing this issue on current master and px4fmuv4.

mcsauder, 2018-04-21:
Hi there,
I am also still seeing this behavior on today's master branch on px4fmu-v4_default build for a quad_x airframe.  I've run through the available commands and the breakage seems to affect all aspects of the pwm module.

The behavior I observe when running the example in the file "pwm test -c 13 -p 1200" is one motor occasionally, sporadically twitching as though one command came through then the next command nulled out the command. In the off chance that somewhere timings have changed, I've tried playing with the delay at line [721 of pwm.cpp](https://github.com/PX4/Firmware/blob/4be4b9810e83f48218c59d5b3cc9dfea575610e9/src/systemcmds/pwm/pwm.cpp#L721) without effect.

I've ensured mc_pos_control and mc_att_control have been stopped, running top verifies this:` PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                   19819 46.211   500/  748   0 (  0)  READY  3
   1 hpwork                        841  1.875   712/ 1780 249 (249)  w:sig 13
   2 lpwork                        222  0.150   656/ 1780  50 ( 50)  w:sig  9
   3 init                         5570  0.000  1608/ 2580 100 (100)  w:sem  3
 490 top                            86  3.225  1296/ 1684 100 (100)  RUN    3
 283 sensors                      3225  3.375  1312/ 1964 249 (249)  w:sem 14
 239 gps                           109  0.225  1048/ 1580 220 (220)  w:sig  5
 285 commander                     418  0.900  2168/ 3212 140 (140)  w:sig 28
 286 commander_low_prio              1  0.000   552/ 2996  50 ( 50)  w:sem 28
 297 fmu                          1000  2.175   760/ 1292 251 (251)  w:sem 10
 300 mavlink_if0                  2429  5.776  1768/ 2540 100 (100)  w:sig 30
 301 mavlink_rcv_if0               240  0.525  1240/ 2836 175 (175)  w:sem 30
 307 mavlink_if1                   871  1.950  1712/ 2564 100 (100)  w:sig 30
 308 mavlink_rcv_if1               222  0.525  1240/ 2836 175 (175)  w:sem 30
 312 frsky_telemetry                 0  0.000   608/ 1188 104 (104)  w:sem  4
 348 mavlink_if2                  4033 10.127  1712/ 2524 100 (100)  w:sig 31
 351 mavlink_rcv_if2               200  0.525  1052/ 2836 175 (175)  w:sem 31
 363 mavlink_if3                  3882  9.227  1768/ 2532 100 (100)  w:sig 31
 366 mavlink_rcv_if3               365  0.450  1552/ 2836 175 (175)  w:sem 31
 409 ekf2                         4255 10.127  4576/ 6572 250 (250)  w:sem 20
 432 navigator                      57  0.150   952/ 1764 105 (105)  w:sem 15
 470 logger                        563  1.200  1032/ 3540 245 (245)  w:sem 36
 471 log_writer_file                 0  0.000   344/ 1068  60 ( 60)  w:sem 36`

Are there any other drivers in this list that might be impacting the pwm module?  Is there any other data I can help collect to continue troubleshooting this module?

Thanks!

-Mark

dagar, 2018-04-21:
At a glance I see that the pwm command is setting the servo output (via IOCTL) directly.

My guess is that this worked in the older releases when px4fmu was only running in HPWORK and the priority of HPWORK was much lower. As a result the `pwm test` would block the regular px4fmu update -> mix -> output loop.

maitham1, 2018-09-04:
Still struggling with this issue, is there anyone that has solved it ?

dgonano, 2018-09-28:
Agreed, wish this was working

bkueng, 2018-10-01:
This should be fixed with https://github.com/PX4/Firmware/pull/10474. Can you test lastest master?

dgonano, 2018-10-01:
@bkueng Yep, working for me now! Thanks

bkueng, 2018-10-02:
Cool! Closing this then.

closed 2018-10-02 by bkueng