PX4/PX4-Autopilot #8852: NAV_CMD_LOITER_TIME_LIMIT never reaches when altitude is different from last waypoint

sanderux, 2018-02-08:
Corner case from actual flight:

WP1:  Waypoint was set at 40m
WP2:  Loiter (time) was set on time: 30 seconds, altitude: 55m, radius: 300m
           The waypoint was placed exactly 300m from WP1
WP3 ...

The vehicle never reached the loiter altitude and never attempted to climb. The waypoint was never considered reached.

Reached check here:
https://github.com/PX4/Firmware/blob/bc3577bd60573b6c1e9f80024659805498e7608b/src/modules/navigator/mission_block.cpp#L215-L216

sanderux, 2018-02-08:
[loiter_time_bug.plan.zip](https://github.com/PX4/Firmware/files/1708818/loiter_time_bug.plan.zip)

dagar, 2018-02-09:
Seems to be FOH, investigating.

dagar, 2018-02-12:
Possibly the same issue in CI (intermittently).
https://logs.px4.io/plot_app?log=4053cbc8-81bb-4f8f-b872-7955398454ac
http://ci.px4.io/blue/organizations/jenkins/Firmware/detail/PR-8874/2/pipeline/728/

closed 2019-02-11 by stale[bot]