PX4/PX4-Autopilot #8853: VTOL 1.7.3: There is no MPC_DEC_HOR_MAX parameter is there other name?

tubeme, 2018-02-08:
Hi guys,

This parameter MPC_DEC_HOR_MAX is very important for us and our frame. Is there name change in the parameters or some other way to control this in 1.7.3?

RomanBapst, 2018-02-13:
I changed the title because this is not just relevant for VTOL. I'll also ping @Stifael and @MaEtUgR here because they will be able to tell you which parameter you should use now.

Stifael, 2018-02-14:
@tubeme what flight behavior did you adjust with this parameter? 
For manual controlled mode, the new parameter is MPC_DEC_HOR_SLOW which controls how fast the vehicle will decelerate when lower speed is required.

tubeme, 2018-02-17:
@Stifael We are using it in the back transition in our VTOL. The VTOL comes in pretty high velocity before the transition. During the transition  the VTOL lowers its velocity but at the end of the transition there is still high velocity. So after the VTOL is in MC and the transition is passed the MC makes a sharp pitch up kick to lower the velocity bellow MPC_DEC_HOR_MAX. Otherwise this pitch up kick almost brakes the vtol. 

Is the MPC_DEC_HOR_SLOW exactly the same as MPC_DEC_HOR_MAX in that regard?

Stifael, 2018-02-19:
The transition is done in auto mode? 
There is no smoothing parameter for auto anymore because for auto related maneuvers since the state of the vehicle should be known and therefore setpoints can be generated directly for a specific state. The state for the vtol transition is known as well, and hence it should be straight forward to decelerate at a certain speed. 
The MPC_DEC_HOR_MAX was never really designed for vtol transition, and it is a surprise to me that it worked especially knowing that the setpoints get reset https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L3136-L3143

tubeme, 2018-02-19:
@Stifael  Probably I did not explained it well...

The problem occurs right AFTER the transition. When the aircraft is in MC mode. In MC mode when the aircraft is moving faster than MPC_DEC_HOR_MAX then the PX4 will try to lower the speed of the MC bellow MPC_DEC_HOR_MAX and it makes this pitch up breaking maneuver.

It does not happen during the transition.

tubeme, 2018-02-19:
Thanks @Stifael I did not explained it correct.

The problem occurs AFTER the transition. And it is MC related. 

In order to have a smoother control in Position mode in MC in 1.6.5 we preset:

MPC_ACC_HOR_MAX = 2
MPC_DEC_HOR_MAX = 2
MPC_XY_CRUISE = 5
MPC_XY_VEL_MAX = 6

The problem is that just AFTER the transition right after the craft switches in MC mode the velocity of the aircraft is about 10 m/s

So the PX4 commands Deceleration of the MC with the pitch up maneuver and its force depends on the MPC_DEC_HOR_MAX parameter. If we want a gentle pitch up (to slow the craft) we need lower value for the DEC parameter.

In 1.7.3 things changed so we could not control this declaration thus could not smooth the maneuver.  

I noticed there is a whole new way and set of parameters to control the MC Position controller in 1.7.3.

MPC_JERK_MAX
MPC_JERK_MIN
MPC_ACC_HOR
MPC_CRUISE_90
MPC_DEC_HOR_SLOW

And we have to somehow smooth the flight just after it entered MC mode in order to NOT have this sharp pitch up.

Is it better explained?

Stifael, 2018-02-20:
I think I understand.
Some of the parameters you mentioned above are only for manual controlled mode, and others for both. Therefore the set of tuning parameters are not necessarily available during the transition.
MPC_DEC_HOR_SLOW has the same purpose as MPC_DEC_HOR_MAX, which is to a parameter meant for controlling the deceleration during manual controlled mode and is part of the slewrate implementation. There is, however, no slewrate for auto anymore because it just screws up the tracking performance. 
Nonetheless, if this is an issue in auto, then it should be straight forward to ramp down the velocity setpoint during transition

tubeme, 2018-02-20:
Thanks a lot @Stifael 

I'm a little confused... By Manual controlled modes you mean all the modes that have pilot's input (Manual, Stabalized, Position, Altitude) or just Manual modes (Ratitude, Stabalized, Acro)?

The MPC_DEC_HOR_MAX is part of the Position controller right? And in our tests it only concerns the Assisted modes ie Position and Altitude as well as fully autonomous modes. In our understanding in the Manual modes only the attitude controller works so MPC_DEC_HOR_MAX, MPC_ACC_HOR_MAX does nothing for the manual modes?

Now with the 1.6.5 we pretty much made the machine to fly very gently and politely in MC Assisted modes. Going to 1.7 we again see these frantic sudden corrections here and there in some situations like just after the transition. So we try to find our way in the new logic in the 1.7 so to smoothen the flight in MC Position and Altitude which in term translated in the same smoothness in the Mission and auto modes in 1.6.5.

What do you mean by "ramp down the velocity setpoint during transition"? how do we ramp down the set point?

Stifael, 2018-02-21:
With manual controlled mode I mean any mode that is controlled by sticks. 
Any parameter with the prefix MPC corresponds to position controlled mode, which can be both manual controlled or auto controlled. The parameter MPC_DEC_HOR_SLOW is, however, only used for manual controlled mode and has no effect during any auto maneuvers. 

With ramping down during transition I mean that the velocity setpoint does not a step from 10 to 0, but rather linearly decreases from 0 to 10.

Stifael, 2018-02-27:
@tubeme did you use the MPC_DEC_HOR_MAX during manual (stick controlled) transitions?

tubeme, 2018-02-27:
@Stifael I understand now. That was my perception that MPC is the position controller and it concerns only the Assisted Manual controlled modes ie Position and Altitude. Thanks a lot @Stifael.

closed 2018-02-27 by tubeme