PX4/PX4-Autopilot #8856: Mapping and RC channel to RSSI

nanthony21, 2018-02-09:
The documentation here: https://pixhawk.org/users/rc_rssi
mentions being able to map an RC channel to the RSSI value using the RC_RSSI_PWM_CHAN parameter. However that parameter no longer seems to exist (I can't find it at least). Is there still a way to do this.

dagar, 2018-02-09:
RC_RSSI_PWM_CHAN still exists, but only on boards with a px4io (pixhawk, pixhawk 2, pixhawk 3). https://docs.px4.io/en/advanced_config/parameter_reference.html#RC_RSSI_PWM_CHAN

Which autopilot do you have?

hamishwillee, 2018-02-09:
@dagar @LorenzMeier There is no other doc of RSSI in the documentation. Is the doc @nanthony21 refers to a reasonable one to include in our docs? (https://pixhawk.org/users/rc_rssi)

Possibly in [Advanced Configuration](https://docs.px4.io/en/advanced_config/)?

nanthony21, 2018-02-09:
I'm using a pixracer. Is there any reason why this parameter shouldn't be available on all boards?

dagar, 2018-02-09:
It was only implemented in the px4io driver (handles communication to the 2nd MCU in a pixhawk). https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io.cpp#L1817-L1823

I don't see why it couldn't be generalized for other boards if this feature is important.

@nanthony21 if you'd like to move this to become a more general feature I could help find an appropriate home.

nanthony21, 2018-02-09:
Sounds good. Let me know what a good place for it would be and I'll take a look at moving it.

dagar, 2018-02-10:
Depending on the board RC input comes from different places. On a pixhawk it's coming in through the px4io, on a pixracer it's read directly in the px4fmu module. It's then published in the input_rc message, which is read by the sensors module to apply RC calibration, and publish a manual_control_setpoint (used by the controllers).

The sensors module sounds like a good place to do the generic RSSI PWM mapping. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/rc_update.cpp#L261

nanthony21, 2018-02-11:
I have this moved over to rc_update and I've tested that it's working on a pixracer. However there are some things that I might need some help with that will need to be taken care of before this will really be usable.

The rssi value coming from rc_update is contained in the `rc_channels` uorb topic but this topic is neither logged nor transmitted on mavlink. Instead the logger and mavlink are using the `input_rc` topic which is coming directly from the platform specific code (px4io, fmu, etc.)

The `input_rc` and `rc_channels` topics contain a lot of duplicate information, would it make sense to combine them into a single topic?

As a side note, I notice that `rc_channels` has a field called `timestamp_last_valid` which is never used.

dagar, 2018-02-11:
I'm not entirely sure why rc_channels is still needed. It's generated along with the other uORB messages, but doesn't seem to be used over uORB.

The other thing to do is make sure you move the corresponding parameter(s)(https://github.com/PX4/Firmware/blob/master/src/drivers/px4io/px4io_params.c#L276) to the same module you use them in.

LorenzMeier, 2018-02-11:
@dagar There is a 3-stage pipeline: 1) Raw RC measurements 2) Calibrated channels 3) manual control channels which are mapped. All 3 stages have value.

dagar, 2018-02-11:
I was mistaken, it's still published. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/rc_update.cpp#L357

nanthony21, 2018-02-11:
So if I'm not mistaken stage 1 of the pipeline is published in the platform specific driver while stages 2 and 3 come from `rc_update`. Rather than logging and transmitting the messages from stage 1, would it make sense to instead use the messages from stage 2? All we would  need to do for backwards compatibility is add the raw RC values to the stage 2 message. Also, I think that there could be a benefit to having access to the calibrated RC values. For example, I know that right now the code that QGroundControl uses to monitor the radio channels just uses the raw values and assumes that the max/min is 2200/800.

dagar, 2018-02-11:
1. input_rc
   - raw RC input hardware dependent (data comes from px4io, px4fmu, or other driver)
2. rc_channels
   - apply calibration and mapping to input_rc
3. manual_control_setpoint
   - map calibrated RC (rc_channels) to normalized manual control input (manual_control_setpoint)

Steps 2 and 3 happen in the sensors module.

nanthony21, 2018-02-11:
A simpler option may be to add RSSI to the manual control set point message.

dagar, 2018-02-11:
I don't think that level should know any of the specifics. Input from mavlink (eg QGC gamepad) comes in directly as mavlink MANUAL_CONTROL -> uORB manual_control_setpoint.

One idea would be to extract the chunk of RC RSSI code and parameters into a small library, then call it from each input module (px4io and px4fmu would be enough for now).

nanthony21, 2018-02-11:
I have it now so that upon receiving new raw data from the `input_rc` topic, `rc_update.cpp` publishes a news topic, `input_rc_out`, with the raw data but modifies the RSSI value. It is then this new topic that gets logged and transmitted over mavlink.

From what I've tested so far it works perfectly with QGroundcontrol. The only issue I see is that because the values are now logged as `input_rc_out` rather than `input_rc` "Flight Review" is not picking up on the logged RC values.

nanthony21, 2018-02-13:
See  PR #8876

closed 2019-02-11 by stale[bot]