PX4/PX4-Autopilot #8858: Changes in px4io firmware needs a "make distclean" to take effect

RomanBapst, 2018-02-09:
@dagar FYI
I was trying to do some debugging in px4iofirmware and the changes didn't seem to have any effect, even though the files were compiled (confirmed by initial compilation errors).
It seems like the binary is not copied into the build directory.
After a **make distclean** the changes took effect.

dagar, 2018-02-09:
The px4io is now treated like a regular first class board and can be built standalone.
    `make px4io-v2_default`

When building a board that uses the px4io interface driver (fmu-v2, fmu-v4pro, fmu-v5) it calls out to build the the px4io binary and copies it into the ROMFS. The dependencies for the standalone px4io-v2_default build should be solid, but I'm guessing the link between the parent (eg fmu-v2) is weak.

One quick solution would be adding a dependency in the top level makefile so that px4fmu-v2_default has a dependency on px4io-v2_default.

dagar, 2018-02-09:
@RomanBapst the dependencies seem to be working in a few random cases I tried. Do you recall which files?

[image]

dagar, 2018-02-09:
It's the FMU side ROMFS. There's no dependency causing the firmware to rebuild after the IO binary is updated and copied into the ROMFS tmp.

PR #8860 (2018-02-09): cmake improve ROMFS dependencies for px4io inclusion
 - fixes #8858
commit f7a17ad5 (2018-02-09): cmake improve ROMFS dependencies for px4io inclusion (#8860)

- fixes #8858
closed 2018-02-09 by dagar