PX4/PX4-Autopilot #8859: replay broken on master

BazookaJoe1900, 2018-02-09:
posix_sitl_replay build doesn't work.
Should it?
is there manual of how to use it? I could only find data about [system_wide_replay](https://dev.px4.io/en/debug/system_wide_replay.html)

dagar, 2018-02-09:
I believe it's done with the regular posix_sitl_default (with an environment variable) set and posix_sitl_replay can be deleted.

[image]

I'll verify and delete.

hamishwillee, 2018-02-11:
@dagar So [system wide replay](https://dev.px4.io/en/debug/system_wide_replay.html) is correct way to do replay right? (I'm just checking this is correct, and all you're doing is deleting the unsupported build config?)

dagar, 2018-02-12:
I don't know the full history, but that appears to be the case. We should add something basic for replay in CI, but as an example, and to keep it working. We also need to make more of an effort to clean up as things evolve, otherwise it's very hard to follow.

bkueng, 2018-02-12:
> So system wide replay is correct way to do replay right?

Yes.

`posix_sitl_replay` is/was used in combination with sdlog2 (which is why we still kept it, but we can start to remove it now).

BazookaJoe1900, 2018-12-14:
This can be closed according to answers

closed 2018-12-14 by BazookaJoe1900