PX4/PX4-Autopilot #8861: Removing trims from FW Manual/Direct Control

tstastny, 2018-02-09:
I noticed the same actuator trim parameters are applied to every control mode, including "manual/direct" mode. I guess the operational concept here would be that one is expected to zero out trims on the remote once a set of trims are found. However, I think this is not the best approach for the following reasons:

1) this makes a MANUAL_OVERRIDE in the case of FMU crash not seamless (would be an actuator jump) from "manual FMU" to "manual pass through" - could confuse the safety pilot, possibly making a dangerous scenario.

2) pilots often re-trim throughout manual flight. these re-trimmed values are then included in the sum for every other (possibly already tuned) non-manual mode.

3) one pilot's trimming may differ from anothers - and resetting controller trims every flight is not ideal.

I would propose that a safer operational concept would have no trims applied in the manual mode (i.e. truly pass through at all times), and the pilot is able to set whatever trims he/she is comfortable with on the remote itself. This way MANUAL_OVERRIDE situations have seamless switching for the pilot. Then, to ensure the stabilized modes are unaffected, it would be the responsibility of the user to make every non manual mode **on the remote with all trims set to zero - this is simple to do.

It would be a quite small change to the fw attitude controller, basically just deleting the trims from the sum in [this line](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L1194).

If this is agreeable - I could make a PR in the next few days.

dagar, 2018-02-09:
We need to think about this overall, but I'd really like to avoid more complexity on the transmitter side if possible.

If we did this then transmitter setup with different trims per flight mode will fall on PX4 documentation. Yes you could say, "please refer to your transmitter documentation", but the reality is it will still need to be explained by us.

Then next thing that concerns me is not wanting to be dependant on actual RC. Many users already want to move between manual control via RC input and gamepad or joystick input over mavlink.

dagar, 2018-02-09:
1. the IO should be using the same TRIM parameters. If that's not the case we should fix it.
2. maybe there's a way we could accommodate this in flight, but still store the actual trim values vehicle side. 
3. I don't see how this is a problem. There's nothing preventing a particular pilot from adding their own additional trimming on top of the vehicle baseline.

tstastny, 2018-02-09:
1. ok, it looks like it does set these trims to the IO, i'll check on the bench to confirm sometime this weekend
2. i guess this is somewhat an intractable problem on the firmware side, as i don't think it is possible to know what remote trims are set - but it has been this way for some time and no one has made any issue, so perhaps whoever does want to avoid the extra trims (from the remote) going on top of the parameter trims in stabilized modes can just take it upon themselves to set the mode dependencies on their own remote - or they would need to go the other way with zero trims on remote.
3. if the trims are set to the IO - sure, agreed. there would be no issue. only the same concern from pt. 2, but as said, pt.2 can be handled then individually on the remote if the user chooses (does not require firmware change).

thanks for the quick feedback and clarification. again, i'll check on the IO trim setting and get back to you - but otherwise it seems we could leave this as it is.

dagar, 2018-02-09:
2. this might be a terrible idea, but the first thing that came to mind is configuring another RC switch that would allow you to immediately set new trims from the current position. There's also simple auto trimming that we could consider.

tstastny, 2018-02-12:
Hi - double checked the MANUAL_OVERRIDE trim writing as promised. It does write to the IO, so no issue with actuator jumps here. 

On 2., I don't think setting new trims from the remote in stabilized mode is really the goal (though could possibly be a cool idea for making initial controller tuning a bit more streamlined). Ideally, one would set parameter trims that are optimal for the controller, then leave them. Then, I agree with your earlier comment that if it's a fixed parameter trim on manual (and also on pass through - as I just double checked) - it is ok for the pilot to just trim as desired around this fixed trim (in the manual mode).

The only concern is when going into stabilized modes - these transmitter trims deviating from the fixed (already optimized/tuned) parameter trims will then be seen as e.g. attitude command offsets in the stabilized mode - e.g. could translate to something like a 5 deg roll command. This is not so nice, but can be mitigated by putting the responsibility on the pilot to configure his/her transmitter to zero out any trims when moving to stabilized modes (again - all on the remote). 

If people only used RC remotes, I would argue against the current approach, but as you mention there are joysticks, game pads, etc to consider - I can understand how the current approach is likely the only way to deal with this. Might be good to put a note in whatever dev wiki page that any additional transmitter trims set in manual **will be put through to stabilized modes if the transmitter is not configured as I mentioned.