PX4/PX4-Autopilot #8862: [RFC] Documentation of build system

jlecoeur, 2018-02-09:
I would like to start a discussion on how to improve the documentation of the build system. I do not have all the knowledge required to make this doc, hence this post.

I believe the following questions are not answered in the doc (please let me know if they are!):
- when you type `make posix_sitl_default gazebo_iris_none` what is the meaning of each element (posix, iris, etc...)
- what are the associated source files, cmake files, init scripts, world files, ...
- what changes are required to customize the build system (for example if I want to add a target, or use a custom init file, or use a custom gazebo model)

Answering these questions would
1. make you more confident when typing a build command by knowing what is actually happening
2. when customizing the build system, it would save hours of browsing through the code / global searching for existing targets

FYI @hamishwillee @dagar 
 
---
Below is what I understood so far, please correct & complete.

### Format of build command
```
make XXX_YYY_ZZZ AAA_BBB_CCC
```
where
- `XXX_YYY_ZZZ` is what is **buillt**.
- `AAA_BBB_CCC` is what is **launched**

example:
```
make posix_sitl_default gazebo_iris_none
```

### Detail of first group ("what is built")
- XXX: **architecture** (ex: posix, nuttx, ...)
- YYY: **config** (example sitl, px4fmu-v2, px4fmu-v3)
- ZZZ: sub-config? (default, lpe)

All associated cmake files are stored flat in the same directory. For example for nuttx_px4fmu-v2_default the cmake file is *cmake/configs/nuttx_px4fmu-v2_default.cmake*.

### Detail of second group ("what is launched")
- AAA: **viewer** (ex gazebo, jmavsim, replay)
    * I haven't looked at it yet
- BBB: **model**, i.e. init & simulation world (ex iris_opt_flow)
    * associated init script (example for iris, the init script is *posix-config/SITL/ekf2/iris*)
    * associated simulation world file (example for iris_opt_flow, the world file is *Tools/sitl_gazebo/worlds/iris*)
    * associated cmake (the name of BBB must be added to the list of **models** in *src/firmware/posix/sitl_target.cmake*)
- CCC: **debugger**
    * I haven't looked at it yet

dagar, 2018-02-09:
This is definitely needed, I just hope we could first remove some of the unnecessary complexity rather than further entrenching it with high quality documentation. Hiding simple things like launching px4, px4 under gdb, or gazebo behind multiple layers of scripting has got to stop. Make simple things simple.

 - I think we need to completely extract Gazebo and the models from the PX4 Firmware build system. There's nothing shared in the build of PX4 Firmware and sitl_gazebo. Previous discussion. https://github.com/PX4/Firmware/issues/6389#issuecomment-351247999
 - this overlaps with the ideas we've had for consolidating all board pieces into single directories. https://github.com/PX4/Firmware/pull/8777#issuecomment-364172322
 - other minor things
   - why don't we don't default from the target names? The vast majority of what people actually use and interact with are "default". Let's starting assuming _default is the default.

jlecoeur, 2018-02-09:
I agree that making the system simpler is a better solution. However even a minimal documentation of the format of the make commands would help a lot.
Regarding the defaults, I would argue that while documentation is missing it is better to be explicit. For example, I did not know that you could add another parameter after "gazebo" before I tried to add my own model. Similarly, I learned about the possibility to debug in gdb just now.

dagar, 2018-02-09:
What I personally want is to not have to know about the majority of these parameters. 

For example if px4 launch was trivial (./px4) and you actually saw it, then the extra magic parameter for gdb (___gdb) wouldn't even need to exist, because it would be obvious (gdb ./px4). Things became unnecessarily complex early on, and instead of rethinking it, even more complexity was piled on top to make it feel simple. Now we have a situation where extremely capable developers can barely figure out how to launch px4 under a debugger.

hamishwillee, 2018-02-11:
1. This is to some extent covered in simulation here: https://dev.px4.io/en/simulation/#startingbuilding-sitl-simulation but it could be a lot more generic because it only applies to posix/sitl. You're also correct that we need more very high level conceptual docs on what the build targets are.
1. That said, I would very much prefer not to invest in this if we're going to be changing the build system soon ... and I think that we should. 

@dagar Where is fixing up the PX4 launch in your "roadmap"? If it something in the coming months or is it "another year".