PX4/PX4-Autopilot #8863: Velocity setpoint jump in auto mode

RomanBapst, 2018-02-09:
@Stifael @MaEtUgR FYI
I found that the copter was a bit too aggressive during an auto mission although the max acceleration was relatively low (2 m/s/s).
Looking at the log I found that the velocity setpoint was not smooth (see the jumps).
Is there a setting I need to change or is this bug?

[image]

In manual mode I found it quite smooth as was the velocity setpoint in the log.

Stifael, 2018-02-09:
@RomanBapst there is no smoothing auto anymore. There is a velocity feedforward along track and position setpoint perpendicular to track, although the feedforward term is incorporated into the position setpoint due to legacy reason. 
The logic is quite simple now. The vehicle always just follow the previous-current line, where it accelerates with MPC_ACC_HOR. The speed at the current waypoint is defined by the angle previous-current-next. 
Can you send the entire log? Did you change MPC_ACC_HOR_MAX or MPC_ACC_HOR?

RomanBapst, 2018-11-06:
Fixed in #10746

closed 2018-11-06 by RomanBapst