PX4/PX4-Autopilot #8864: VTOL Flight Issue

LorenzMeier, 2018-02-09:
@RomanBapst Could you please have a look at this log file?
https://review.px4.io/plot_app?log=6fa2399a-ccb4-4a2d-8303-091dec28dda0

@PX4/testflights You can tag Roman for future issues directly.

r0gelion, 2018-02-09:
@LorenzMeier for VTOL issues or any issue?

LorenzMeier, 2018-02-09:
Any controls related issue.

RomanBapst, 2018-02-11:
@r0gelion Hi, I was looking at the above logs From what I can see the vehicle started to roll over immediately after the transition. Was this vehicle flying before or was this a maiden flight? I'm asking because it looks like the direction of the elevons was reversed or something like that.
If this was not the case let me know and I'll dig deeper.

RomanBapst, 2018-02-11:
@r0gelion Also I cannot see in the logs that the vehicle transitioned back to copter mode. It did stall at some point but until the very end there was not transition, as suggested in the description in the log.

dagar, 2018-02-11:
Is there any video of this flight?

Do you have any preflight checks for fixed wing or vtol?

santiago3dr, 2018-02-13:
Last successful flights on this vtol:

vtol take-off and landing; good tracking in auto:
https://logs.px4.io/plot_app?log=7857f742-6dc4-45f2-83f9-3b8ac64c2d4b
https://logs.px4.io/plot_app?log=fb6061ba-31a9-4a64-a097-8af3489bdc16

manual take-off and transition to fw; stable flight:
https://logs.px4.io/plot_app?log=d749fb0e-a3cf-4d3d-a072-0d55c3b5d22e

RomanBapst, 2018-02-14:
@santiago3dr @dagar My conclusion here is the following:
1) You entered a spin by pulling up too much and thus reducing airspeed.
If you have a look at the animation you really get that impression
https://logs.px4.io/3d?log=6fa2399a-ccb4-4a2d-8303-091dec28dda0

2) Something went wrong with the elevons (failure).

I cannot completely rule out the second point but my feeling is that 1) was most likely to happen.

RomanBapst, 2018-02-14:
@santiago3dr By the way, I don't know if you are aware of that but you can enable quadchute based on euler angle violation. You can read more about it [here](https://dev.px4.io/en/advanced/parameter_reference.html), search for **VT_FW_QC_P**
I think this feature would have saved your wing in this case.

santiago3dr, 2018-02-14:
The odd flight that had a back momentary back transition to MR and then back to FW for a rough landing:
https://review.px4.io/plot_app?log=63b57036-d008-4917-b170-f17b8d142f8d

@dagar you suggested to check possible rc loss

LorenzMeier, 2018-02-25:
@RomanBapst Let's try to find a final answer here.

RomanBapst, 2018-02-26:
@santiago3dr @LorenzMeier I had a look at the last log which you posted.
The  'unwanted' backtransition was a result of the vehicle receiving a DO-TRANSITON command when it was flying in AUTO mode. From the vehicle command fields it looks like the transition was part of the mission it was running. The vehicle did the transition upon which the pilot switched to stabilized mode. Since the transition switch was held in the ON position (fw flight) the vehicle transitioned to forward flight again. After that the vehicle remained in stabilized mode until it landed.

So far everything makes sense and I don't see any bug.
@santiago3dr Is there any other issue you want to raise from this? the only thing I could think of is to change the handling of the transition switch, e.g. do not transition if came back from AUTO mode and first require the user to flip the switch into the  mode(hover or forward flight) the vehicle currently is in.

I will close this for now, please reopen if you think there is an problem we should fix.

closed 2018-02-26 by RomanBapst

santiago3dr, 2018-02-26:
That all makes sense, the transition switch was on MR by default as both transition to FW and rtl were part of the waypoint mission; I will try to adjust the mission and transition manually prior to Auto so that any back-transitions do not come as a surprise.