PX4/PX4-Autopilot #8866: Mavlink command routing to companion computer

myeabhishek, 2018-02-10:
I want to MAV_CMD mavlink messages coming from Ground control system to my companion computer connected to Pixhawk via telem 2. Basically I want to handle some of the commands on ROS for example camera controlling commands. 

Can you please tell me how can I route mavlink messages coming from QGC to companion computer or if it has been done already?

CarlOlsson, 2018-02-11:
Start the mavlink app with forwarding active (-f flag) in order to route messages between QGC and an on board computer via the autopilot.

hamishwillee, 2018-02-11:
On most pixhawk boards this is automatically enabled using SYS_COMPANION -
 https://dev.px4.io/en/companion_computer/pixhawk_companion.html#pixhawk-setup

I "think" the code for this is in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS

myeabhishek, 2018-02-12:
So the forwarding is enabled now. Can you please tell me under which rostopic will I get MAV_CMD data?

hamishwillee, 2018-02-12:
I can't. I suggest you raise question on the discussion forums.

lamping7, 2018-02-14:
There is no ROS topic for this. https://github.com/mavlink/mavros/issues/943

LorenzMeier, 2018-02-25:
I'm closing this - please use http://discuss.px4.io for discussions. Thanks!

closed 2018-02-25 by LorenzMeier