PX4/PX4-Autopilot #8867: PX4 compile on Ubuntu wants to use nonexistent python install

nanthony21, 2018-02-10:
I'm trying to build the firmware on Ubuntu but I kept getting errors about missing python modules. Even though typing "python" in the terminal defaults the the system's python2.7 for some reason PX4 is trying to use my Anaconda install of python3.6. I tried deleting Anaconda all together but now when I try to build I get an error saying that `~/anaconda3/bin/python` cannot be found. How can I get make to stop trying to use the wrong version of python?

Here is the traceback that I get:

`[6/867] Generating uORB topic headers
FAILED: cd /home/nick/Firmware/msg && /home/nick/anaconda3/bin/python tools/px_generate_uorb_topic_files.py --headers -f actuator_armed.msg actuator_controls.msg actuator_direct.msg actuator_outputs.msg adc_report.msg airspeed.msg att_pos_mocap.msg battery_status.msg camera_capture.msg camera_trigger.msg collision_report.msg commander_state.msg cpuload.msg debug_key_value.msg debug_value.msg debug_vect.msg differential_pressure.msg distance_sensor.msg ekf2_innovations.msg ekf2_timestamps.msg esc_report.msg esc_status.msg estimator_status.msg follow_target.msg fw_pos_ctrl_status.msg geofence_result.msg gps_dump.msg gps_inject_data.msg home_position.msg input_rc.msg irlock_report.msg landing_target_innovations.msg landing_target_pose.msg led_control.msg log_message.msg manual_control_setpoint.msg mavlink_log.msg mission.msg mission_result.msg mount_orientation.msg multirotor_motor_limits.msg offboard_control_mode.msg optical_flow.msg parameter_update.msg position_setpoint.msg position_setpoint_triplet.msg power_button_state.msg pwm_input.msg qshell_req.msg rate_ctrl_status.msg rc_channels.msg rc_parameter_map.msg safety.msg satellite_info.msg sensor_accel.msg sensor_baro.msg sensor_bias.msg sensor_combined.msg sensor_correction.msg sensor_gyro.msg sensor_mag.msg sensor_preflight.msg sensor_selection.msg servorail_status.msg subsystem_info.msg system_power.msg task_stack_info.msg tecs_status.msg telemetry_status.msg test_motor.msg time_offset.msg transponder_report.msg tune_control.msg uavcan_parameter_request.msg uavcan_parameter_value.msg ulog_stream.msg ulog_stream_ack.msg vehicle_attitude.msg vehicle_attitude_setpoint.msg vehicle_command.msg vehicle_command_ack.msg vehicle_control_mode.msg vehicle_global_position.msg vehicle_gps_position.msg vehicle_land_detected.msg vehicle_local_position.msg vehicle_local_position_setpoint.msg vehicle_rates_setpoint.msg vehicle_roi.msg vehicle_status.msg vehicle_status_flags.msg vtol_vehicle_status.msg wind_estimate.msg -i /home/nick/Firmware/msg -o /home/nick/Firmware/build/px4fmu-v4_default/uORB/topics -e templates/uorb -t /home/nick/Firmware/build/px4fmu-v4_default/msg/tmp/headers -q
/bin/sh: 1: /home/nick/anaconda3/bin/python: not found
[6/867] Generating mixer_multirotor_normalized.generated.h
FAILED: cd /home/nick/Firmware/build/px4fmu-v4_default/src/lib/mixer && /home/nick/anaconda3/bin/python /home/nick/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --normalize -f /home/nick/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_t.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_x.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_h.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_y.toml /home/nick/Firmware/src/lib/mixer/geometries/tri_y.toml /home/nick/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
/bin/sh: 1: /home/nick/anaconda3/bin/python: not found
[6/867] Generating mixer_multirotor.generated.h
FAILED: cd /home/nick/Firmware/build/px4fmu-v4_default/src/lib/mixer && /home/nick/anaconda3/bin/python /home/nick/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py -f /home/nick/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_t.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_x.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_h.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_y.toml /home/nick/Firmware/src/lib/mixer/geometries/tri_y.toml /home/nick/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
/bin/sh: 1: /home/nick/anaconda3/bin/python: not found
[6/867] Generating mixer_multirotor_6dof.generated.h
FAILED: cd /home/nick/Firmware/build/px4fmu-v4_default/src/lib/mixer && /home/nick/anaconda3/bin/python /home/nick/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/nick/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_t.toml /home/nick/Firmware/src/lib/mixer/geometries/hex_x.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/octa_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_h.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/nick/Firmware/src/lib/mixer/geometries/quad_y.toml /home/nick/Firmware/src/lib/mixer/geometries/tri_y.toml /home/nick/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
/bin/sh: 1: /home/nick/anaconda3/bin/python: not found
[6/867] Generating apps/builtin/registry/px4_iridiumsbd_main.bdat
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v4_default' failed
make: *** [px4fmu-v4_default] Error 1`

dagar, 2018-02-10:
It's not setup to specifically to use pythonin anaconda, so you'll either need to change your path or specifically point PX4 at the desired python.

Can you run `which python` and `export | grep -i python`?

With any changes stashed run `make distclean`, then `make` again, but this time save the entire output (I'm looking for the initial cmake configure steps).

nanthony21, 2018-02-10:
`which python` returns `usr/bin/python` and `export | grep - i python` is blank. I'm not sure what changed but this time the build worked fine, thanks for the help.

closed 2018-02-10 by nanthony21