PX4/PX4-Autopilot #8872: RC transmitter must be on when PX4 starts

nanthony21, 2018-02-12:
If the RC transmitter is turned on after the flight controller has already initialized then RC will not be used for control setpoints. It seems that this issue was introduced between 1.5.5 and 1.6.0rc1.

I think that it is because there are two instances of the manual_control_setpoint topic being published, one for RC and one for SOURCE_MAVLINK_3. If the RC transmitter is on during startup then it's manual_control_setpoint topic will be instance 0, but If it is turned on after startup then the mavlink instance will be instance 0 and the RC instance will be instance 1.

dagar, 2018-02-12:
This was actually a short term "feature" to prevent them from conflicting. Ideally we'd want some hand off mechanism to move between RC and gamepad input, but it needs to be safe.

nanthony21, 2018-02-12:
Thanks, that makes sense.  I must've had virtual joystick enabled on QGC.  Making sure that it is disabled seems to prevent this issue.

closed 2018-02-12 by nanthony21