PX4/PX4-Autopilot #8885: Smooth takeoff in auto considerations

RomanBapst, 2018-02-14:
@Stifael @MaEtUgR We tried auto takeoff in mission mode and get extremely aggressive behavior (throttle jumping to hover thrust).
I noticed two things that can go wrong:

1) As far as I understand the z velocity setpoint is ramped up from negtive values to positive values in order to achieve a thrust ramp. I found that the takeoff velocity limit was hardcoded [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L3150). I wonder if this could lead to problems since I assume that this gives a dependency on your velocity controller gains?

2) The vehicle which we are using (Matrice 600) takes a bit more time to arm (large propellers) and as a result the takeoff begins before the vehicle is able to takeoff. As a result you get a huge jump in thrust because the controller starts to ramp up before the vehicle is armed. You can also reproduce this in SITL with jmavsim BTW. I think we should add a bit of a delay to prevent this scenario. As a workaround I manually armed the vehicle and then put it into mission mode.

After using the workaround in 2)  we got really nice takeoff behavior.

Let me know what you guys think.

Stifael, 2018-02-14:
@RomanBapst did you also try to adjust MPC_TKO_RAMP_T? This values controls how fast the velocity should ramp up https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2533

RomanBapst, 2018-02-14:
@Stifael Yes, that works fine. The essential issue is really just that the controller starts the ramp when the vehicle is not fully armed yet. You can reduce the effect by making the ramp slower but that is not the proper fix. The system should just wait a couple of seconds after arming before it starts the takeoff.
I think we used to have a delay in there.

dagar, 2018-02-14:
What if we had a ramp lower level regardless (the mixers)? That would directly account for arming and not allow immediate jumps straight to hover. You'd still want roughly the same logic higher level, but this would then apply to all modes.

MaEtUgR, 2018-02-21:
@RomanBapst 1. You're right I'm aware of this. It nevertheless worked better than without ramp in every scenario I tested. I planned to further improve the ramp and making it lower level and thought we could ramp the thrust for idle speed again instead of starting with a negative velocity but make sure the slope and stopping criterion results in the exact same it is right now because it works very good for manual position mode takeoffs with different "agressivity" on the joystick.
2. That's a case we never had because the vehicles we tested all arm super fast. Could we make a delay available but configurable?

@dagar I think the mixer is not the problem because the ESC is basically what can not execute your command from the beginning and without you knowing. Best would be to have feedback of the ESC if the motor is already on idle speed and ready but that's not generally available.

albert21martins, 2018-04-12:
Hi,
I have a pixhawk 2.1 & Here GPS within a 5kg Copter. When I tried to takeoff from QgroundControl one of the motors burned. Not happening when I do it from the RC. 
I armed the motors before init takeoff from QGC in oder that the motors already turn. Then I realize once I started the Takeoff, the motors drove down and then suddenly full gas. This burned several motors, while changing several parameters.
tried MPC_TKO_RAMP_T, 1.5

I am afraid to test it again, any idea?

MaEtUgR, 2018-04-12:
@albert21martins That sounds severe 😲 If motors are able to burn on takeoff it's very likely that your ESC - motor hardware setup is not optimally layed out. Nevertheless we could have a look at the actual output that gets sent to the motors and if the software assists the hardware to reach its limits. Can you download the logs from your vehicle and post the one that is working manually and the one which didn't?

deksprime, 2018-10-31:
Hello. Currently we are having a similar problem with a 780 mm quad which weighs at about 5 kg. Here are our logs from different takeoff attempts. Here are our logs with many different values of `MPC_TKO_RAMP` where I think that this parameter isn't good for anything, it only changes the ramp after the copter starts to hover. Here are our logs with `MPC_TKO_RAMP` mostly constantly increased and there's no pattern of dropping. The copter takes off at about 40A when I takeoff by hand but sometimes reaches 120A when taking off in auto mode. 

Surroundrone 780 Mission Test 1 => https://review.px4.io/plot_app?log=7efb52bb-a543-4ab8-99f2-04d38e303926
Surroundrone 780 Mission Test 2 => https://review.px4.io/plot_app?log=68af5941-5504-48d9-bbce-413408889d78
Surroundrone 780 Takeoff Test 1 => https://review.px4.io/plot_app?log=27f29f62-d4e0-4137-850c-01b7a4d466f1
Surroundrone 780 Takeoff Test 2 => https://review.px4.io/plot_app?log=c921db3f-dde0-4536-a522-11dde2cf8cdd
Surroundrone 780 Takeoff Test 3 => https://review.px4.io/plot_app?log=5e1589b4-3f87-4a6c-a0a5-255b14aa71c0
Surroundrone 780 Takeoff Test 4 => https://review.px4.io/plot_app?log=f60a21fb-2cf0-47ff-beae-d9f32028db3c
Surroundrone 780 Takeoff Test 5 => https://review.px4.io/plot_app?log=bcc49517-a821-4e0f-bb50-1874ccfa41aa
Surroundrone  780 Takeoff Test 6 => https://review.px4.io/plot_app?log=e577df61-c07d-4b93-a2ab-89f09e780f29
Surroundrone 780 Takeoff Test 9 => https://review.px4.io/plot_app?log=24b40994-feb5-4303-a1c1-bd73a827534a

bresch, 2018-11-02:
After merging #10746 we should be able to refactor properly the takeoff logic that is in `mc_pos_control_main.cpp`. The ramp will then be automatically generated by the flight task.

closed 2019-02-18 by stale[bot]