PX4/PX4-Autopilot #8891: uORB replace all IOCTL usage

dagar, 2018-02-15:
We need a uORB mechanism to completely replace the use of IOCTLs for sensor drivers. The main thing is device id, and calibration get/set. The rest are nearly unused outside of the driver itself (a few small exceptions).

If we move this to uORB the vast majority of sensor drivers wouldn't actually need any IOCTLs or even a CDev.

@LorenzMeier thoughts?

dagar, 2018-02-15:
Do we need a ROS services equivalent? http://wiki.ros.org/Services
@RomanBapst

LorenzMeier, 2018-02-25:
@dagar Yep.

dagar, 2018-04-04:
@thomasgubler @bkueng @RomanBapst any thoughts on this?

It's also an opportunity to recover quite a lot of flash. The majority of IOCTL code is manually duplicated, including across very similar devices.

bkueng, 2018-04-05:
I agree, a ROS services equivalent would be benefitial.

dagar, 2019-01-31:
Moving to v1.10.0

dagar, 2019-06-25:
Slowing getting there.

dagar, 2020-04-08:
Almost there. Should be completely done by v1.12.0.
https://github.com/PX4/Firmware/pull/14237

dagar, 2020-09-06:
Done!

closed 2020-09-06 by dagar