PX4/PX4-Autopilot #8897: Px4 does not fly along straight line between waypoints

jdzerve, 2018-02-16:
**Bug Report**

PX4 does not fly along straight line between waypoints.

Scenario:
Create mission with 2 waypoints:
1st WP at altitude 10m;
Set speed after WP1 to 1m/s;
2nd WP 30m apart with altitude difference 30m;

Set MPC_Z_VEL_MAX_UP and MPC_Z_VEL_MAX_DN to 5 m/s

Expected behavior:
Copter flies along straight line from WP1 to WP2 with ~1 m/s ground speed.

Actual behavior:
After reaching 1st WP copter flies up rapidly at ~5m/s and then horizontally to the next WP at ~1 m/s ground speed.

This could be ok in most scenarios but in the opposite direction (when descending rapidly before flying horizontally) is not ok at all. One could tune the MPC_Z_VEL_MAX* params to achieve straight flight but it does not work for missions with various slopes.

dagar, 2018-02-17:
Could you share a log? https://logs.px4.io/

VertLim, 2018-03-23:
Any news on this issue? I experienced exactly the same behaviour yesterday using the 1.7.3 release.

VertLim, 2018-04-11:
@dagar Will this issue maybe be fixed with release 1.7.4? Sorry for asking again and many thanks for your reply.

closed 2019-02-11 by stale[bot]