PX4/PX4-Autopilot #8901: uorb_graph is missing some connection

BazookaJoe1900, 2018-02-17:
Some publishers are missing from the graph, its happens when the module uses 'uORB::Publication'
For examples:
vehicle_local_position publisher form ekf2
https://github.com/PX4/Firmware/blob/f932217fdc706e5ac02769b41444e4e9a7e3e4a4/src/modules/ekf2/ekf2_main.cpp#L336

I have noticed to few more, but not much.
I guess that for subscribers it happens too with 'uORB::Subscription'

hamishwillee, 2018-02-19:
@bkueng Assigned to you because I assumed this was about https://dev.px4.io/en/middleware/uorb_graph.html ...  but it might not be.

bkueng, 2018-02-19:
I'm aware of that, and as I wrote in https://github.com/PX4/Firmware/pull/8393, I planned to do some refactoring for these classes, before adding them to the graph.

hamishwillee, 2019-01-29:
@bkueng Assume you are still keeping an eye on this one.

bkueng, 2019-01-29:
Yes

closed 2019-12-21 by BazookaJoe1900