PX4/PX4-Autopilot #8902: FW actuator_controls_2 parachute interferes with gimbal (vmount)

dagar, 2018-02-17:
[image]

https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L872

dagar, 2018-02-17:
At a minimum the parachute needs to be moved to not interfere with the gimbal.

dagar, 2018-02-17:
Is anyone actually using parachute on control group 2?

LorenzMeier, 2018-02-17:
Right now I think nobody is really using it.

dagar, 2018-02-17:
https://github.com/PX4/Firmware/pull/8907

sanderux, 2019-09-18:
DO_SET_SERVO is publishing on this too:
https://github.com/PX4/Firmware/blob/8a8cc3b52b45c12078c7c655c3b29c6f5511b7f6/src/modules/navigator/mission_block.h#L138
https://github.com/PX4/Firmware/blob/8a8cc3b52b45c12078c7c655c3b29c6f5511b7f6/src/modules/navigator/mission_block.cpp#L449

julianoes, 2019-09-20:
FYI @bresch

dayjaby, 2019-10-05:
Having the same problem now too. Controlling gimbal with AUX channels does not work for fixed wing/VTOL. How expensive would it be to create non-interfering control groups? Like one for only gimbal. One for external servos. And control group 6 for parachute. This would give a lot of freedom if you only have to create custom mixer files.

This could be also used to cleanup a few parameters. For example "TRIG_PINS" could be handled by a camera control group. RC_MAP_AUX# could be handled by a RC passthrough control group.

ryanjAA, 2019-10-22:
Having the same issue as well.