PX4/PX4-Autopilot #8906: VTOL: RTL problem.

tubeme, 2018-02-17:
Hi guys,

In the current implementation or Return (RTL) for VTOL, if we are at large altitude 300-500-1000m and we activate RTL then the VTOL performs airplane RTL at the current altitude until it reaches the home spot, and imminently makes the transition. Then it starts to lower the altitude in MC mode down to the RTL altitude.

If we are at 500m height this is very unpractical energy wise. Because if we have to go down from 300-1000m to 100m with MC it is very very slow and the battery drains in a seconds especially with the presumption that this happens at the end of flight. 

More practical would be to have the RTL implemented like in the FW flight. It comes back at current altitude. Then it makes a spiral down to the RTL height (100m) and then makes the transition to MC.

We've had a flight at 500m and the return almost killed our machine while descending in MC.

dagar, 2018-02-17:
You'd need to extend RTL here to descend before transition. https://github.com/PX4/Firmware/blob/master/src/modules/navigator/rtl.cpp#L212

Then there's the question of how transition fits into it. Ideally I think you'd want to reposition and transition towards home.

hamishwillee, 2018-02-19:
Note, if a change gets made here it will impact this doc: https://docs.px4.io/en/flight_modes/rtl.html

rollysing, 2018-02-22:
I'm not a coder but might I present a programming consideration when dealing with this but from the perspective of a full scale pilot. During cross country flights or with airliners, you choose a cruising altitude that would give you best efficiency for fuel and time consumption. In the flight planning stage, you determine a TOD (top of decent) waypoint which is where you start descending from your cruise altitude at a desired descend rate to your pattern altitude to land at the airport. The main factor is groundspeed.

To make it for simple calculation and assuming no obstacles en route, if you're last waypoint is 3000 ft from Home, cruising at 1100 ft with a groundspeed of 500 ft/min and your target RTL altitude is 100 ft at Home waypoint, with a descent rate of 500 ft/min and a transition time of four seconds from FW to MC your TOD is 1033 ft from Home.

Hope that made sense and sparks some thought process. Maybe it can be made as an optional behavior of VTOL RTL.

sanderux, 2018-02-22:
@dagar, should we always loiter down to descend to RTL altitude before transitioning?

rollysing, 2018-02-23:
Would it be possible to provide the option to either Loiter and spiral down to RTL altitude or straight-in descent to Transition then Land? Just a personal preference for me as a full scale pilot and when considering a means to an end, I'd rather be straight and level than banking, especially when wind strength and direction comes in the formula. Just a thought...

sanderux, 2018-03-01:
We know the max descent speed for the FW, why not stay at current altitude as long as possible and descend at the lates moment to reach RTL altitude?

tubeme, 2018-03-09:
We think that energy wise it is much better to spiral down, because banking in the spiral does not require motor, it is rather gliding down. We also could release the battery for a moment, while glyding down, before the back transition starts. Which is punishing the battery at that time at the end of the flight. So the less we spend in vertical descend the better energy wise. It is tested by us for our configuration. So may be a parameter to switch between both is good idea. In strong weather the quad drains the battery even faster, so again it is better to spiral down. The plane can cope with the weather well tuned.

rollysing, 2018-03-09:
Please, take me as a devil's advocate presenting this and served as food for thought.

Your reasoning is very valid but couldn't you accomplish the same thing from straight-in approach? Your theory works well on a no-wind scenario but I'm not sure how well, even a fine tuned aircraft, will respond when you add the wind component. I've attached the effect of wind for illustration purposes. 

I'm presuming that the flight controller, during the spiral gliding descent will not only maintain airspeed but also equidistant at all quadrants of the spiral against the home point. Assuming the wind is from the north, the aircraft will encounter its steepest bank on the east quadrant of the turn. Not only that but it also has to pitch down to maintain the airspeed. Depending on the wind speed, it wouldn't be possible that it will reach the max bank angle parameters. If that happens, then it'll fly a bit farther away from the home point? Or will the FW motor give bursts of power to maintain the symmetrical circle?
[image]

tubeme, 2018-03-09:
It copes believe me. :) It is funny sometimes to watch but it manages pretty well to spiral down even in strong weather. The thing is we make the loiter radius a little bigger so to lower the bank angle. And we limit the bank angle as well. 

Ineresting implementation would be once the vtol reaches the RTL altitude to make another two circles sense the wind direction exit from the loiter from the correct angle and make the transition against the wind... :)

rollysing, 2018-03-09:
Oh good. Do you set the loiter radius in the params or is that something you can setup when creating the mission?

tubeme, 2018-03-09:
It is in the parameters. In Mission page NAV_LOITER_RAD

rollysing, 2018-05-30:
Just curious if there was any implementation of this or when it might be considered?

sanderux, 2018-05-31:
on it

moreba1, 2018-10-10:
I have an idea for RTL. This idea needs to change in autopilot code and the Qgroundcontrol software . we can design a landing pattern such as a fixed wing landing pattern using the software. The UAV must returns to a loiter(altitude) waypoint in FW mode ,Then when it’s reach to direction of home, it must at that altitude start the transition to to MC and reach to the top of the home. Then reduce the altitude to Land .
In this landing pattern we can define this parameters:
1- distance(to Land point) and heading of loiter waypoint
2- loiter altitude
3- landing waypoint

moreba1, 2018-10-12:
what is your opinion؟  @tubeme  @dagar @sanderux

closed 2019-02-04 by stale[bot]