PX4/PX4-Autopilot #8911: Actuator Control logging frequency in System Identification mode seems too small

vmishra9, 2018-02-18:
Hi! I am trying to perform a realistic system identification using **actuator_controls_0_0** and **sensor_combined_0** data so that I can get a respectable PID gains using Matlab. 

I set the SDLOG_PROFILE to 3 which is the System Identification mode but on analyzing the data I can see that actuator_controls_0_0 is recorded at only 10 Hz (approx) whereas sensors_combined_0 is logged at 250 Hz (approx). I know that the attitude control runs at a minimum of 50 Hz. Is there parameter to increase the logging frequency of actuator controller? Ideally, can I log all the actuator control values through some parameter?

dagar, 2018-02-18:
Did you reboot after changing SDLOG_PROFILE?

vmishra9, 2018-02-18:
Thanks for the reply @dagar . SDLOG_PROFILE = 3 (System Identification mode) is the default setting.

dagar, 2018-02-19:
@vmishra9 SDLOG_PROFILE is now a bitmask.

3 = default + ekf replay
[image]

Enabling system identification would be 11.
[image]

vmishra9, 2018-02-19:
ok. Changed SDLOG_PROFILE to 11 and now I have both sensors_combined and actuator_controls at 250 Hz (approx.)

One more question though, in sensors_combined, is **gyro_rad[i]** data angular velocity or delta angle ? I am confused because it has a name gyro_rad and not gyro_rad_s.

dagar, 2018-02-19:
It's angular velocity (averaged over gyro_integral_dt).

closed 2019-02-09 by stale[bot]

tecdee, 2019-07-23:
@neom89 Pls how did you carry out your system identification? I have a custom drone which I have built using px4. I am trying to obtain a model so that I can design more precise PID gains. Please help. Thanks.