PX4/PX4-Autopilot #8913: Hardfault after attitude PID tuning flight

deksprime, 2018-02-18:
I have come across a hardfault as shared in this fault log. Before this log, I have been extensively tuning the rollrate PID's. I was changing the `MC_ROLLRATE_P` parameter while the copter was on flight but I got this hard fault after the flight was done. I suspect that the issue might be related to changing parameters around 10 times or so during the flight but I haven't got any direct proof of this. I'm using Pixhawk the Cube as the flight controller and testing on 1.7.3 stable.
```
[hardfault_log] -- 2018-02-18-10:39:20 Begin Fault Log --
System fault Occurred on: 2018-02-18-10:39:20
 Type:Hard Fault in file:semaphore/sem_post.c at line: 118 running task: mc_att_control
 FW git-hash: 50bd148f53f1aeca6f1bd4a1caabcc18e4f2888b
 Build datetime: Jan  6 2018 11:32:25
 Build url: localhost 
 Processor registers: from 0x2001ab28
 r0:0x00000002 r1:0x2001ab28  r2:0x20008278  r3:0x00000000  r4:0x20020f94  r5:0x00000010 r6:0x2001aab0 r7:0x00000000
 r8:0x20020f94 r9:0x20003174 r10:0x00000064 r11:0x00000000 r12:0x0801fca5  sp:0x20020f50 lr:0x0800741f pc:0x08011188
 xpsr:0x21000000 basepri:0x00000010 control:0x00000004
 exe return:0xffffffe9
 IRQ stack: 
  top:    0x200028f0
  sp:     0x20002870 Valid
  bottom: 0x20002604
  size:   0x000002ec
  used:   000000c8
 User stack: 
  top:    0x20021068
  sp:     0x20020f50 Valid
  bottom: 0x200209dc
  size:   0x0000068c
  used:   0000068c
Interrupt sp memory region, stack pointer lies within stack
0x20002921 0x200132c0
0x20002920 0x000000b1
0x2000291f 0x2000d2b0
0x2000291e 0x0000ffff
0x2000291d 0x00000000
0x2000291c 0x0000ffff
0x2000291b 0x00000000
0x2000291a 0x0000ffff
0x20002919 0x00000000
0x20002918 0x0000ffff
0x20002917 0x00000000
0x20002916 0x0000ffff
0x20002915 0x00000000
0x20002914 0x000000ab
0x20002913 0x2000f8e0
0x20002912 0x000000aa
0x20002911 0x10008490
0x20002910 0x00000149
0x2000290f 0x20022df0
0x2000290e 0x0000ffff
0x2000290d 0x00000000
0x2000290c 0x000000e7
0x2000290b 0x20014bd0
0x2000290a 0x000000e6
0x20002909 0x20014e60
0x20002908 0x0000ffff
0x20002907 0x00000000
0x20002906 0x00000144
0x20002905 0x20019840
0x20002904 0x00000003
0x20002903 0x10002620
0x20002902 0x00000002
0x20002901 0x10001740
0x20002900 0x00000001
0x200028ff 0x10000860
0x200028fe 0x00000000
0x200028fd 0x20002bcc
0x200028fc 0x2000aef0
0x200028fb 0x20008200
0x200028fa 0x00000000
0x200028f9 0x00000000
0x200028f8 0x00000000
0x200028f7 0x00000000
0x200028f6 0x00000000
[... 677 lines cut ...]
0x20020eb1 0xb9999fc5
0x20020eb0 0xbc9f4f98
0x20020eaf 0xb9999fc5
0x20020eae 0x3f7ff39b
0x20020ead 0x3f7fe4aa
0x20020eac 0xbc91ec48
0x20020eab 0xbcba4018
0x20020eaa 0xbce7bd6a
0x20020ea9 0xbee3d1b6
0x20020ea8 0xbf652588
0x20020ea7 0x3bbef836
0x20020ea6 0x3f653736
0x20020ea5 0xbee3fb6c
0x20020ea4 0x00000000
0x20020ea3 0x0807c51b
0x20020ea2 0x3f06d10f
0x20020ea1 0x20020d38
0x20020ea0 0x2002024c
0x20020e9f 0x20020f64
0x20020e9e 0x00000000
0x20020e9d 0x00000000
0x20020e9c 0x00000000
0x20020e9b 0x00000000
[hardfault_log] -- 2018-02-18-10:39:20 END Fault Log --
```

M-Skelton, 2018-02-21:
I saw the same hard fault in FW

[image]

Antiheavy, 2018-02-22:
If it is any help getting to the bottom of this, here is a link to a fixed wing log with this same hard fault "type".  No idea what happened, but we are very interested in finding out why:
https://logs.px4.io/plot_app?log=fcb05575-9317-49d7-932a-31aaf9dccd7d

Antiheavy, 2018-02-22:
Here is another fixed wing hard fault log, different hard fault "type" though.  may or may-not be related to running into some trees...  Chicken-or-egg?  Not sure what happened first, the hard fault or the tree:
https://review.px4.io/plot_app?log=5a04b85e-dac6-4404-885f-2aa7cb529482

dagar, 2018-02-22:
Two of these, (@M-Skelton and one from @M-Skelton) are in the fmu when running as a task.

@M-Skelton which board are you using?

M-Skelton, 2018-02-23:
@dagar I was using an fmuV4 board.

bkueng, 2018-02-23:
@deksprime did you flash the version via QGC? If not, can you provide the binary that you flashed?

The only log that points to a valid upstream commit comes from @deksprime. Nevetheless it helps of course collecting different incidents, but we need to make sure we're not chasing different things.

Here's how to debug hardfaults: https://dev.px4.io/en/debug/gdb_debugging.html#debugging-hard-faults-in-nuttx

deksprime, 2018-02-23:
@bkueng The version was flashed through the QGC choosing the Stable firmware option

bkueng, 2018-02-26:
Thanks.
Can you provide the following:
- What hardware you have attached to the board and to which connector(s)?
- The list of configured parameters (an ulog file should do it)?

bkueng, 2018-02-27:
Let's collect what we know:
- RAM usage is below critical levels
- no low-stack warnings in the logs
- the hardfault happens on different vehicle types (fixed-wing & multi-copter)
- the hardfault happens on different boards (pixhawk cube & Pixracer)

Looking only at the initial report (which runs stable w/o additional changes, but the other logs show the same hardfault):
- the hardfault happens in IRQ context (and I think the user stack is not relevant)
- it's an assertion failure in `semaphore/sem_post.c`: `ASSERT(sem->semcount < SEM_VALUE_MAX);`, with `SEM_VALUE_MAX = 0x7FFF`
- the IRQ stacktrace is:
```
(gdb) info line *0x08005939
Line 356 of "armv7-m/up_assert.c" starts at address 0x8005938 <up_assert+420>
   and ends at 0x8005946 <up_assert+434>.
(gdb) info line *0x080b7fa0
No line number information available for address 0x80b7fa0
(gdb) info line *0x08011188
Line 94 of "armv7-m/gnu/up_switchcontext.S"
   starts at address 0x8011188 <up_switchcontext+10> and ends at 0x801118a.
(gdb) info line *0x080074db
Line 119 of "semaphore/sem_post.c" starts at address 0x80074da <sem_post+38>
   and ends at 0x80074de <sem_post+42>.
(gdb) info line *0x08009e21
Line 925 of "../../src/drivers/stm32/drv_hrt.c"
   starts at address 0x8009e20 <hrt_tim_isr+132>
   and ends at 0x8009e2a <hrt_tim_isr+142>.
(gdb) info line *0x08009d9d
Line 609 of "../../src/drivers/stm32/drv_hrt.c"
   starts at address 0x8009d9c <hrt_tim_isr> and ends at 0x8009da0 <hrt_tim_isr+4>
```

The assertion failure can be due to 2 causes:
- a memory corruption somewhere in the codebase (including NuttX)
- the semaphore counter overflows due to an unbalanced use of the semaphore

The only place matching this stacktrace is the logger, scheduling its main loop (there's a similar stack trace pattern in uORB, but there's a condition that excludes it as a candidate).
The semaphore in the logger can indeed overflow, if the hrt keeps firing, but the logger's main loop is blocked for some reason (e.g. a higher-prio task running busy, or a problem in the logger itself). For the counter to overflow, the logger must be blocked for at least almost 2 minutes.

https://github.com/PX4/Firmware/pull/8979 fixes the semaphore counter overflow, but **not** the root cause here.

---

@dagar can we have the `.elf` file available on S3 right next to the `.px4` file? This will help with debugging such cases, as we can be sure the binary matches the release (I had to download the same toolchain as CI uses for this now).

---

Did anyone of you notice if the QGC (mavlink) connection got lost about 2 minutes before the hardfault happened?

Antiheavy, 2018-02-27:
> Did anyone of you notice if the QGC (mavlink) connection got lost about 2 minutes before the hardfault happened?

I'm not the original poster, but interestingly in our flight the QGC MAVlink telemtry .TLOG went for some seconds longer than the SDcard.  Screen shots below and .TLOG attached in the ZIP folder.

[2018-01-25 15-26-01.zip](https://github.com/PX4/Firmware/files/1764755/2018-01-25.15-26-01.zip)
.ULOG:

[image]
[image]

deksprime, 2018-02-27:
Unfortunately, I wasn't able to track what was happening on the QGC. I haven't been able to find the SD card of the copter in which this hardfault has happened so I didn't share a flight log. I will share a flight log right away if I find the SD card which I hope might eliminate some other possible causes.

bkueng, 2018-02-28:
Thanks for the info. Based on this, it's more likely a memory corruption. And to narrow it down further, I will need to know which drivers & modules are/were running.

Can you provide me with the following:
- What hardware you have attached to the board and to which connector(s)?
-  The list of configured parameters (an ulog file started after bootup should do it)? The logs from above have logging since boot activated, and thus do not contain all params.

Antiheavy, 2018-02-28:
With regards to this log:
https://logs.px4.io/plot_app?log=fcb05575-9317-49d7-932a-31aaf9dccd7d

Hardware:
- FMUv4, custom layout
- mRo GPS002-MR, Tx/Rx connected to the GPS connector, powered externally
https://store.mrobotics.io/mRo-GPS-u-Blox-Neo-M8N-HMC5983-Compass-p/gps002-mr.htm 
- ms4525 diff pressure with HMC5983, SCL and SDA on i2c connector, powered externally
- Telem 1 and Telem 2 Tx/Rx to two different "companion computers"

We have custom init scripts to set parameters, load drivers, etc., I've attached a zip folder with our custom scripts that should show all that.  
[PX4-init-scripts-master.zip](https://github.com/PX4/Firmware/files/1768134/PX4-init-scripts-master.zip)
There might be some small differences in parameters between this zip folder and 
what was flown in the log, but I think nothing that would affect sensors, hardware, or logging.

Antiheavy, 2018-03-07:
To add another datapoint to this investigation, we've been going through old SDcard logs and came across another semaphore/sem_post.c Hardfault on a different vehicle.  This must have happened during ground handling, not a flight.  Identical hardware setup to our other fixed wing hard fault logs.  Should be the same init scripts as the ZIP folder I attached previously.
https://review.px4.io/plot_app?log=21d2c119-3aa8-4881-96f3-d11d8b01e380

bkueng, 2018-03-09:
Interesting, this is quite a while back (Jul 14 2017 20:37:28), and it shows the same pattern.

I did not see anything special from the startup scripts & params, the setup and configuration does not look out of the ordinary.

Antiheavy, 2018-03-13:
Here is another semaphore hardfault.  Same configuration as the other logs I've post previously.  This was from a while back.  Just adding to the issue here in case someone ever finds something in the future.

Log:
https://review.px4.io/plot_app?log=3d54eac4-996d-4590-89ad-708ab64bd2eb

[image]

LorenzMeier, 2018-03-13:
@bkueng Could you have a look? @thomasgubler FYI

bkueng, 2018-03-14:
It looks like the same issue. A noticeable difference is that there are no param changes in the latest report, whereas in most previous logs there were. And there is an airspeed sensor error: `[cal] Airspeed sensor is reporting errors (260)`.

I still see the following potential causes:
- some high-prio task starts to busy-loop
- a memory corruption somewhere in the code-base (NuttX or PX4)

julianoes, 2019-07-02:
@davids5 does this still need following up?

davids5, 2019-07-02:
@julianoes - yes someone should repeat the test with capture and see if the problem still exists.

julianoes, 2019-10-08:
@dinomani is this something you could try at some point?

bresch, 2020-01-24:
We were not able to reproduce this issue in 2 years, closing.

closed 2020-01-24 by bresch