PX4/PX4-Autopilot #8918: SYS_COMPANION - update speed

ShikOfTheRa, 2018-02-19:
Hi, I m experiencing slow update of some data on OSD. Is this adjustable without re-compiling?

In particular GPS_RAW_INT which is sent from PX4 only once every 5 seconds. 
It makes some OSD items look poor. Especially distance from home and distance to waypoint. 

I looked in code and can see some values which I thought would be the refresh rate against each mavlink command. In reality it appears they are running at a rate of 20% of that value for me. I put some diagnostic code on OSD to determine update frequency and it verifies refresh speeds are all approx 20% of the values in link below: Maybe this is a task scheduler process to minimise CPU / serial performance? 

https://github.com/PX4/Firmware/blob/e04d30c9cf8d02603b85bdffd13c844bea773265/src/modules/mavlink/mavlink_main.cpp#L2062

Example Video (apologies for quality - just a test flight...):
https://www.youtube.com/watch?v=k7E9gu-OtRY

dagar, 2018-02-19:
Can you confirm which SYS_COMPANION setting you're using?

With everything connected could you run `mavlink status` on the console? 
https://dev.px4.io/en/debug/system_console.html
Alternatively you can do this through QGC.

Which OSD hardware do you have?

ShikOfTheRa, 2018-02-19:
Thanks for reply

Am using 157600
Genuine PIXRACER
OSD is stock minim
Seems to work OK with arducopter. In line with the SR hz parameters  

I put some debug on OSD to  capture the Hz of a couple of commands. 
VFR_HUD is running @5 - its fine at that. 
GPS_RAW is 0.2 - or every 5 secs

I just tried playing a TLOG captured from someone else. I presume its Ardu of some description and seeing 35hz/8hz for the commands above so the OSD appears to process OK. 

What would I look for in QGC?

closed 2018-03-21 by ShikOfTheRa