PX4/PX4-Autopilot #8919: VTOL: Descending back transition problem.

tubeme, 2018-02-19:
Hello guys,

We perform this descending transitions with our VTOL. while pitching down for some time loosing altitude, we engage the transition switch.

Because the Baro sensor is laggy and GPS is laggy as well we have a situation where the craft thinks it is sinking while it is not (seen from the ACC data) and it performs full ramp up (to the max allowed MC thorttle) for a couple of seconds until the baro and GPS catch up. This makes the VTOL climb a lot 5-10m+ during the back transition.

Here are three flights with different OMEGA:

FW_T_HGT_OMEGA = 1
Which should give almost all the wight towards the ACC data.

https://logs.px4.io/plot_app?log=137c1889-8e43-4595-9640-505dfeadce9b

FW_T_HGT_OMEGA = 5

https://logs.px4.io/plot_app?log=29582da5-77d0-4697-a74f-32a11ff1455b

FW_T_HGT_OMEGA = 10

https://logs.px4.io/plot_app?log=572d6bac-dc04-4e99-8578-7dbae23dd20d

And there is almost no difference. 

We want to give even more weight on the ACC data in the fusion but there is no way it can be done.

May be a solution should be if the transition switch is enabled to have the VTOL leveled for 1-2 seconds (ignoring he manual input) and then to make the MC ramp up? 

For 1-2 seconds the sensors should settle the altitude and remove the problem.

sanderux, 2018-02-22:
@tubeme your back transition duration is much too little, the quadcopter goes full throttle to brake the vehicle because it exceeds MPC_XY_CRUISE. You need to start applying active breaking and increase your backtransition duration. 

Also i am not sure that questions about tuning your vehicle belong in the github issues list. You might try any of the community channels available such as the forum or slack chat

closed 2018-02-22 by sanderux

tubeme, 2018-02-23:
Probably you are mixing our problems because of the other posts.

This is not the case where we have a very short transition and thus the Position controller in MC mode breaks the speed.

This is different. If we have a plain level flight transition there is no such an excessive throttle during the back transition ramp up. During level flight the transition is flawless.

The problem occurs when we have a descending transition. And not just a small descend but rather longer descend.

sanderux, 2018-02-23:
The problem is the same, 4 seconds is not enough time to burn enough speed to get the vehicle at the correct speed. If you apply active braking it will go away