PX4/PX4-Autopilot #8925: [SITL crash] px4 node crashes on startup

TobiMiller, 2018-02-20:
Hey,

i can successfully compile with "make posix_sitl_default gazebo". the px4 node crashes when i run "roslaunch px4 mavros_posix_sitl.launch". 

Is this a known problem? 

i have compiled protobuf v3 for cartographer. i returned to protobuf v2.6.1 from source because the px4 firmware doesnt compile with v3.  

what exactly means the symlink error?

[image]

lamping7, 2018-02-20:
Delete the three symlinks in the .ros directory that point to the firmware:
1. ROMFS
2. posix-configs
3. test_data

So, `cd /home/tobias/.ros/` then `rm` the three symbolic links
Then try again.

TobiMiller, 2018-02-20:
Thanks a lot. That was the problem. as easy as that.

closed 2018-02-20 by TobiMiller