PX4/PX4-Autopilot #8927: Unstable sensors cause Critical problem

Ahrovan, 2018-02-20:
Bug Report

I used Pixhawk hardware's (listed below), but in Most of the time see "PREFLIGHT FAIL" message for MAG,GYRO,ACCEL and in one case cause crashed.

[log uploaded (Pixhawk LITE 2.4.6)](https://logs.px4.io/plot_app?log=b0be3ceb-c180-40b9-9cc4-d31ef1c1920a)

```
[commander] PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
[commander] PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
[commander] PREFLIGHT FAIL: MAG SENSORS INCONSISTENT
[commander] PREFLIGHT FAIL: MAG #1 UNCALIBRATED
```

 - All boards calibrated with USB cable, in some case with WiFi module

The hardware's list (all with PX4 Firmware v1.7.3 + gps module m8n):

- Pixhawk 2.4.6 (maybe 3DR original version)
- Pixhawk 2.4.6 (Copy version)
- Pixhawk LITE 2.4.6

LorenzMeier, 2018-02-20:
@dagar Is this related to the open detection PR?

Ahrovan, 2018-02-20:
@LorenzMeier Probably a mistake occurred in your reply :)

LorenzMeier, 2018-02-20:
No it's not.

Ahrovan, 2018-02-21:
@dagar @LorenzMeier What is your opinion about this problem? Do you need more information for troubleshooting? I try with another pixhawk and send log file today.

dagar, 2018-02-21:
@Ahrovan it's possible your issue is related to https://github.com/PX4/Firmware/pull/8703, https://github.com/PX4/Firmware/issues/8643. I've discussed with @davids5 and have a potential fix to implement and test later tonight.

Could you expand on exactly what you did? When did you calibrate relative to flashing? How are you powering the pixhawk? Did you command a restart from QGroundControl (including selecting an airframe) or the reboot command in nuttx?

dagar, 2018-03-14:
Could you try https://github.com/PX4/Firmware/pull/9074?

Ahrovan, 2018-12-18:
@dagar The problem persists with the latest version (v1.8.2 Stable Release)

hansbonini, 2019-03-27:
Same problem on 1.9-dev

julianoes, 2019-07-10:
@hansbonini and @Ahrovan could you answer to this:

> Could you expand on exactly what you did? When did you calibrate relative to flashing? How are you powering the pixhawk? Did you command a restart from QGroundControl (including selecting an airframe) or the reboot command in nuttx?

And can you try latest master?

julianoes, 2019-10-08:
Closing because no one followed up.

closed 2019-10-08 by julianoes

Ahrovan, 2019-10-08:
I'm a follower. But there is no feedback from developers

hamishwillee, 2019-10-08:
@Ahrovan This request was waiting on a response here from you or @hansbonini:
 https://github.com/PX4/Firmware/issues/8927#issuecomment-509977006

If you're happy to act as the stakeholder on the reporter side (?) I'm sure @julianoes is more than happy to reopen and continue investigation