PX4/PX4-Autopilot #8928: How to connect mpu9250 to stm32f767zi

99FengQingYang99, 2018-02-20:
Hey, guys!
I am new to PX4. Here is the situation. I have flashed the px4nucleoF767ZI-v1 to stm32f767zi Nucleo board. And I am able to run mpu9250 application in Nutshell. But I don't know which pins on the Nucleo board I should connect to using SPI connection. Could you guys give me some suggestions, please?

ksschwabe, 2018-03-02:
This question should rather be posted here: http://discuss.px4.io/.

I have not tried this myself, but I know where in the code to look to find what pins are configured.

The board initialization files can be found here:  Firmware/src/drivers/boards/px4nucleoF767ZI-v1/
  - Looking at board_config.h, the MPU250 CS line is connected to PF2 (pin define GPIO_SPI_CS_MPU9250)
  - The SPI is initialized in spi.c - the MPU9250 is on bus SPI1.

To find out which pins are configured for MISO, MOSI and CLK on SPI1, you need to look at the NuttX config files. These can be found here:  Firmware/platforms/nuttx/nuttx-configs/px4nucleoF767ZI-v1:
  - In px4nucleoF767ZI-v1/include/board.h, you will find the definitions for GPIO_SPI1_MISO (PA6), GPIO_SPI1_MOSI (PD7) and GPIO_SPI1_SCK (PG11).

I hope that helps you.

99FengQingYang99, 2018-03-03:
Thank you for your response! I connected the MPU9250 as the way you said. And then I get the following:

nsh> mpu9250 test
mpu9250 test
ERROR [mpu9250] bus 0 not started

nsh> mpu9250 -s test
mpu9250 -s test
ERROR [mpu9250] bus 3 not started

Do you know how to start bus 0 or 3?

ksschwabe, 2018-03-04:
Have you started the mpu9250 driver already? You need to run `mpu9250 start` before you can run `mpu9250 test`.

99FengQingYang99, 2018-03-04:
Thank you for your response, ksschwabe! I just run mpu9250 start and then the mpu9250 test. 
I get the following:
nsh> mpu9250 start
mpu9250 start
WARN  [mpu9250] probe failed! 0
WARN  [mpu9250] no device on bus 3

nsh> mpu9250 test
mpu9250 test
ERROR [mpu9250] bus 0 not started

nsh> mpu9250 -s start
mpu9250 -s start
WARN  [mpu9250] probe failed! 0
WARN  [mpu9250] no device on bus 3

nsh> mpu9250 -s test
mpu9250 -s test
ERROR [mpu9250] bus 3 not started

Do you know why probe failed?

99FengQingYang99, 2018-03-04:
Thank you, ksschwabe! I will look into those issues.

99FengQingYang99, 2018-04-18:
Hey, @ksschwabe. 
I have checked the connection over and over. So the connection is correct. But I still can't start the mpu9250.
I used a logic analyzer to check the spi connection and I get the following diagram.
[image]
I have no idea what is wrong. Could you give me some suggestion, please?

ksschwabe, 2018-04-18:
It looks like the MPU9250 is not responding (the MISO line is not responding). Are you sure that the MPU9250 is powered corectly?

99FengQingYang99, 2018-04-18:
Hey, @ksschwabe 
I am sure the MPU9250 is powered correctly. Because the datasheet of MPU9250 showed that for SPI and I2C have the same setting for powering it up. And I have used I2C to write and read the MPU9250.  
I can get all the data from the MPU9250. I used the same setting to power up the MPU9250. 
Do you have any other suggestions?  I am thinking about connecting the MPU9250 with the NUCLEO board using I2C. Could you tell me how to do it?

ksschwabe, 2018-04-19:
I think it is easiest to try and use the I2C_EXPANSION bus (as defined in board_config.h). This is I2C4. I2C is configured with I2C4_SCL on PF14 and I2C4_SDA on PF15.

When starting the MPU9250 driver, I think you can still just use MPU9250 start, otherwise you will need to pass it the 'X' option to start on the external I2C bus, `MPU9250 -X start`.

ksschwabe, 2018-04-20:
Do you have pull-ups on the I2C lines? (Probably not if you are connecting it directly to the Nucleo board. Or maybe your MPU9250 dev board has the pull-ups.)

You could try using I2C2 (SCL -> pin PF1 and SDA -> pin PF0) or I2C1 (SCL -> pin PF1 and SDA -> pin PF0).

99FengQingYang99, 2018-04-20:
Hey @ksschwabe 
I have tried to use the I2C1 and I2C2. And the Nucleo board still gives me nothing.
I am using the Sparkfun MPU9250 breakout board. The board has the pull-up resistors in their SCL and SDA line.
I am afraid the code is not working. Could you give me some suggestions, please?

ksschwabe, 2018-04-21:
Unfortunately I don't have a Nucleo STM32F767ZI board, so I can't test this myself. You will have to dig through the code and try and see where it is failing.

You can try and post your problem on http://discuss.px4.io/. Hopefully someone there will be able to help you.

99FengQingYang99, 2018-04-21:
I will try it. Thank you @ksschwabe!

closed 2018-04-21 by 99FengQingYang99

confidential96, 2018-05-06:
Were you ever able to resolve this? I am having a similar issue.