PX4/PX4-Autopilot #8930: Battery Estimation Initialization on some Platforms like Intel Aero Fails

MaEtUgR, 2018-02-21:
After merging #8153 there seem to be certain platforms on which the initialization of the battery estimate seems to fail. The result is that if you start up the platform even with a full battery you are not allowed to take off because the low battery states are already engaged and you can by definition not get out of them again.

Reported cases are:
- @ndepal Unkown Platform
- @santiago3dr Intel Areo
- @ChristophTobler Intel Aero

I have access to an Intel Aero so I should be able to test. And I already have the suspition that these platforms report a very low/invalid voltage in the beginning and after that it the low battery states can not recover anymore.

santiago3dr, 2018-02-21:
This looks to also have affected the OcPoc
Upon bootup and connection to qgroundcontrol, i get a low/critical battery warning despite battery being fully charged and shown at 100%
[image]

lbegani, 2018-02-22:
I am also facing the same issue with Aero, find the log here - https://logs.px4.io/plot_app?log=ea64f182-b898-4a4b-8f76-85638f99610e

PR #8940 (2018-02-24): Pr aero battery initialization hotfix
On the Intel Aero and probably also other specific platforms the first measured voltage values despite connected battery are unuasable. The solution here is to start filtering and determining warning only after a measurement above 2.1V was received. This condition was anyways used before to determine if the battery status report gets filled at all.

I bench tested this on an Intel Aero. Can someone else please also confirm?

fixes #8930
closed 2018-02-24 by LorenzMeier