PX4/PX4-Autopilot #8931: Snapdragon: broken communication with QGroundControl

santiago3dr, 2018-02-21:
Running current master or stable firmware, snapdragon fc does not fully connect to qgroundcontrol.

I can partially connect to qgcbut get error shown below upon connecting:
[image]

Unable to upload mission file, get this error:
[image]

Have flashed stable firmware on the snapdragon, attempted stable build of qgroundcontrol to no avail.
No sure if this is on the firmware side or qgc

LorenzMeier, 2018-02-26:
@ChristophTobler Do you have this issue on your end as well?

ChristophTobler, 2018-02-26:
Just tried with current master and QGC v3.3.0: It doesn't connect to QGC and the msg is (repeatedly)
```
Info: [pm] unknown param ID: xxxx
```

santiago3dr, 2018-04-09:
I'm not sure if this is firmware relevant but with 1.7.4 beta, and QGroundControl 3.3.1 stable , i am able to connect, calibrate and read/write a mission file.

Having similar issues with the Intel Aero, see #9259

closed 2018-08-01 by santiago3dr