PX4/PX4-Autopilot #8934: Flight Issues

dronecrewkari, 2018-02-22:
@priseborough @jgoppert 

I had a flight today with Q_estimator + LPE today.

with optical flow and Lidar-Lite v3.
(Lidar-Lite v3 was automatically used for LPE)

Every parameter for Q_estimator + LPE was default parameter.

Previous flight with the same UAV for EKF2 was okay as following log indicates a few days ago

https://logs.px4.io/plot_app?log=3027489e-bfee-44b4-9dff-97ed295723b3

However, with Q_estimator + LPE today

I cannot control UAV even in Position Hold mode, so changed into RTL mode immediately.

Following log is based on Q_estimator + LPE.

https://logs.px4.io/plot_app?log=7b7a3db4-3549-46a5-ae63-76a5c918447c

It seems to me that yaw angle estimate was quite strange.

Here is the parameter file for LPE. 

[Flight_with_LPE.tar.gz](https://github.com/dronecrew/kari_ws/files/1746970/Flight_with_LPE.tar.gz)

Would you please let me know what was the problem?

priseborough, 2018-02-23:
@jgoppert are you able to handle this? I'm not familiar with the LPE code.

TSC21, 2018-02-23:
@priseborough I will review this soon.

LorenzMeier, 2018-02-25:
@dronecrewkari The LPE is not community supported, so I'm closing this. Ping us if you need support for it and we can point you at different consultants who might be able to offer support. We as a community can't ensure quality on all sorts of configurations and focus our attention on the officially supported components.

closed 2018-02-25 by LorenzMeier

TSC21, 2018-02-25:
@LorenzMeier I will take it from here.